iNavFlight/inav #1485: Pre arm setup "navigation is safe" is always red

Nexus1212, 2017-03-30:
Hello,
I use the Inav Version 1.6.1 with a SPRacingF3 and a NZ mini GPS. 

In the pre arm setup "navigation is safe" is always red. In the tab GPS is for few seconds my GPS position, then I get a error message.

My problem is I can not arm the fixed wing, if I use a the mode RTH.

Nexus1212, 2017-03-30:
In thread #1276 the same issuewas discribed, but it was closed without solution.

digitalentity, 2017-03-30:
It is NOT safe to arm the machine in RTH, PH or WP. This is intentional.

Nexus1212, 2017-03-30:
RTH is not aktive, if i arm the aircraft. It is only the option available, to activiate RTH via transmitter. I arm in mode Passtrough or acro.

digitalentity, 2017-03-30:
Safety will block arming in any of the following conditions:

1. You configured WP/PH/RTH or Failsafe RTH and you don't have a good accuracy GPS fix
1. You try to arm into RTH/PH/WP
1. You have waypoint mission in memory and your first waypoint is too far from your current position

Given it's a small NZ mini GPS my guess is that accuracy is not good. You need at least 6 sats and position accuracy better than 9 meters (usually HDOP of 5-6 or lower).

Nexus1212, 2017-03-30:
Thank you for your response! 

I had only 5 sats vor gps fix. Do you mean, with 5 sats inav gets no valid GPS fix?

digitalentity, 2017-03-30:
No 5 sats is not enough for good navigation. Even 6 sats is marginal. I consider at least 7-8 satellites acceptable for flight.

I think we can close this now?

Nexus1212, 2017-03-30:
Yes, thank you very much! Do you have a cheap recommendation for a gps?

sppnk, 2017-03-30:
I have five or six of this working all with INAV.
Sometimes I get 19 SATS. Many times I catch sats indoors in my workbench.
I like the engage fix led feature of them.

http://www.banggood.com/1-5Hz-VK2828U7G5LF-TTL-Ublox-GPS-Module-With-Antenna-p-965540.html?rmmds=search

Nexus1212, 2017-03-30:
Ok, I will buy it :) I hope the shipment to Germany will not take so long. 

Thank you

digitalentity, 2017-03-30:
@sppnk what voltage do you supply to that GPS? 5 or 3.3V? I am considering buying a few

sppnk, 2017-03-30:
I am connecting  directly to FLEXIPORT from cc3d, so it is Vcc PWR unregulated. In my case aprox 5 volts, as I use normal servos. I have similar setup with afronaze mini. I dont remember exactly, but I supousse the supply is also 5V.

But VK2828U can be supplied from 3V3 to 5V. See here: https://github.com/CainZ/V.KEL-GPS/blob/master/VK2828U7G5LF%20Data%20Sheet%2020150902.pdf

digitalentity, 2017-03-30:
Great, thanks for the info. Going to buy a few

closed 2017-03-30 by digitalentity

laurentopia, 2019-01-27:
> 
> 
> Safety will block arming in any of the following conditions:
> 
>     1. You configured WP/PH/RTH or Failsafe RTH and you don't have a good accuracy GPS fix
> 
>     2. You try to arm into RTH/PH/WP
> 
>     3. You have waypoint mission in memory and your first waypoint is too far from your current position
> 
> 
> Given it's a small NZ mini GPS my guess is that accuracy is not good. You need at least 6 sats and position accuracy better than 9 meters (usually HDOP of 5-6 or lower).

2. How do I NOT arm into WP/PH/RTH if they are part of the modes setup?
3. how far is too far?
could we get error messages in the cli to indicate which condition triggers the unsafe?
thanks.

Normoa, 2019-02-01:
I'm sat here with my quad connected to INav, and INav showing a 'Navigation is safe' error - so unable to arm it.

9 sats, HDOP 2.4 (it's indoors), Mode is Horizon. INav FC is 2.0.1

This tends to happen often, but I can never get any info as to what the problem is!

petertorelli, 2021-09-06:
Seconding @Normoa ... if navigation is not safe, please explain *why* it is not safe. Some debug assistance would be great UX.

jdrozdz, 2025-05-03:
> Seconding [@Normoa](https://github.com/Normoa) ... if navigation is not safe, please explain _why_ it is not safe. Some debug assistance would be great UX.
That question is for deves. Second, it's ridiculous to force `Navigation is safe` to be one of the parameters to arm. If someone wants to arm without that point, it should be possible; each operator/pilot of UVA is responsible for his machine. It smells like MS rules: ` We know better than the user.`

sensei-hacker, 2025-05-03:
> > Seconding [@Normoa](https://github.com/Normoa) ... if navigation is not safe, please explain _why_ it is not safe. Some debug assistance would be great UX.
> 
> That question is for deves.

Look at the OSD or the status output to see why it is not safe.

> Second, it's ridiculous to force Navigation is safe to be one of the parameters to arm. If someone wants to arm without that point, it should be possible; each operator/pilot of UVA is responsible

Think that through for a second. It's *literally* impossible. Physically impossible for the craft to do what you'd be commanding. Suppose you are in waypoint mission mode. Meaning you are commanding the aircraft to fly to particular GPS coordinate. With no working GPS module.  It's quite impossible to fly to a particular point without a working GPS.  Cruise mode?  The same. It is impossible to fly due west without a working compass or GPS.

In other cases, you would be giving contradictory commands. It's literally impossible to follow contradictory commands. One such case is when you set maximum EPH to 200 (meaning you, the pilot, are telling it to NOT arm with EPH greater than 200), then you flip the Arm switch while EPH is 800. You TOLD it to not arm with an EPH that high. If it armed anyway, it would be doing the opposite of what you commanded it to do.

> BTW, I have the same situation, I cannot arm my drone. I have more than 6 sats, fix type is as 3D and nothing.

Okay. What does the OSD say?  Or 'status'?

jdrozdz, 2025-05-03:
In my case, it was something with the same IDE because when I returned to Betaflight and configured (again), everything was working fine. So I released if in Betaflight I can arm craft than in Inav, so I re-flash FC and configure step by step.

After calibrating the accelerometer and magnetometer, I checked the status tab. All was green, so I saved the settings, and Inav froze. I saw `Waiting for data`. After 4 minutes, I closed the app (disconnecting was not working), and when I ran again, there was an issue with  `Navigation is safe`.

So, again, the same song, flash, configure and at this time, after calibration, Inav loaded data normally, and all marks were green. There is some issue with saving data (**maybe**). I run Inav under Linux.