iNavFlight/inav #1489: Request to add DSHOT

TwoToneEddy, 2017-03-30:
It would be great if DSHOT was implemented in inav. The great thing about inav is the combination of racing performance and GPS flight modes, adding DSHOT would allow for faster loops, especially now F4 processors are available.

digitalentity, 2017-04-03:
Let's keep it towards at least 1.8. I don't see a lot of benefit in DSHOT for many reasons.

vinicentus, 2017-05-29:
#1393

closed 2017-05-30 by digitalentity

RCnator, 2018-03-05:
So, we're at 1.9.0 now. Will we get dshot compatibility in the future? I'd really like not having to do throttle calibration.

aubreyrjones, 2018-05-23:
> Let's keep it towards at least 1.8. I don't see a lot of benefit in DSHOT for many reasons.

I'm just getting back into modern RC after a decade hiatus. I don't see any reason to use software that doesn't support DShot. I've never experienced the nonsense people talk about in forums... calibration and noise-sensitive signals.

Being used to the deficiencies of analog throttle doesn't invalidate the forward progress made elsewhere in the industry.

RCnator, 2018-06-04:
Any chance there will be dshot compatability with 1.9.2?

VoicOfReason, 2018-07-16:
I'd like Dshot 1200. My ESC supports it, but iNav can't utilize it...... yet.

terragady, 2018-07-28:
yeah I hope for dshot too :) 600 would be enough

RCnator, 2018-07-28:
Obviously me too but sadly we know it is 2.0 now, at least 2.0 RC4, and the request was put forth in March 2017 during version 1.6...

ReganGTX, 2018-07-31:
Please!! DSHOT 1200

teckel12, 2018-07-31:
@ReganGTX Because there's really no advantage and a higher likelihood of catastrophic failure with DSHOT. So, there's little desire to work to add something that the programmer will never use, and will cause a flood of "my model fell from the sky and crashed" issues.

It's also not as cool as it sounds, as it's not really a digital transmission. Current sensors in the ESCs would also not be useful to me as I'd just combine the total amperage anyway and I already have current sensors.

I hope this explains (at least from my perspective) why I have zero motivation to add DSHOT support. But, feel free to add it if you think it's of value. I'm sure a working DSHOT pull request will be merged into INAV.

TwoToneEddy, 2018-07-31:
I think D-shot is really worth adding for the following reasons:

No ESC calibration
DShot1200 is faster than multi shot
D-shot commands for turtle mode, ESC beeper etc.

I'm happy to look at adding this in a couple of weeks once I've got my build environment sorted out again

teckel12, 2018-07-31:
@TwoToneEddy Slightly, but not noticeably faster, with high risk of total failure. And ESC calibration takes like 2 seconds.

But seriously, it's not that a PR would be refused, it's just that everyone who actually knows DSHOT isn't interested in using it so there's no desire to do the programming.

ReganGTX, 2018-07-31:
@teckel12 I appreciate the response and I'm sure you are way more knowledgeable when it comes to this kind of stuff but I would like to add that I have been a quad pilot for the past 5 years and have flown with Dshot from day one of its release and I have never had any catastrophic failures flying with it.

TwoToneEddy, 2018-07-31:
Once I've got my Linux machine up and running after I move house in 2 weeks I'll implement this. I'm a big fan of DShot

ReganGTX, 2018-07-31:
@TwoToneEddy SWEET! THANK YOU!

digitalentity, 2018-08-01:
@TwoToneEddy wait a bit. I will add a timer DMA framework after we release 2.0 (similar to what is there in Betaflight) - it will make adding dshot easier

TwoToneEddy, 2018-08-01:
@digitalentity Ok no worries, I haven't looked at the betaflight implementation yet anyway.

davidngrc, 2018-08-02:
+1 for DSHOT support.
No ESC calibration is a huge advantage for beginner or people always forgot.

terragady, 2018-08-02:
@davidngrc not really, calibration is very easy and 1,5 year ago everyone was doing it as there was no choice, also for 1 ESC setups it is a no brainer :) but yes i think it would be nice to have DSHOT.

aubreyrjones, 2018-08-02:
I mean, pumping nitromethanol fuel is very easy and 15 years ago everyone
was doing it as there was no choice.

DShot is pretty obviously the way the industry's headed. Fighting against
it would just make INAV seem antiquated.

teckel12, 2018-08-02:
@aubreyrjones Then add it instead of talking about it on a closed issue. People in the know would never use DSHOT, therefore there's no urgency to add something that will only cause problems, just to look cool.

TwoToneEddy, 2018-08-02:
@teckel12 Why would people not use it? I don't think I've been to a race and seen anyone using anything else for quite a while now, it seems pretty much the standard. Digital wins over analogue when you're trying to convey a digital value as there's no interpretation needed.

terragady, 2018-08-02:
I agree a little bit with @teckel12 , iNav has different purpose than racing in 200m range where liability is not an big issue but timing and precision is. Anyway if someone would add it, it would be just nice but most of the people can live without it.

aubreyrjones, 2018-08-02:
For me the issue is not so much the perceived precision and speed, but
rather advancing feature set. Reversible direction enables turtle mode,
which seems like a (minor) asset in attempting to recover a multirotor lost
somewhere difficult to access. Buzzer commands can permit omission of
discrete buzzer modules. Eventually ESC telemetry can permit monitoring of
per-motor current and measured RPM (something I'm personally concerned with
while tuning a differential-thrust wing).

That said, I've also had my share of issues reliably using those features
on Betaflight craft, so I can see the more conservative point of view. But
new ESC features require expanded bandwidth on the signal line beyond what
a unidirectional PWM pulse can do.

teckel12, 2018-08-02:
@TwoToneEddy DSHOT may be fine for small racing drones where crashes are common and wires are short.  But for larger craft over longer distances (which is INAV's wheelhouse) DSHOT can cause catastrophic situations.  Also, DSHOT is **NOT** digital over the wire, it's still analog.  Digital is also not always better, that's marketing brainwashing.  How do your digital speakers or headphones sound?

So, there's no urgency to add a feature that's mostly not needed and could be quite dangerous or cause a flood of issues.  If any of you in this thread want to program DSHOT, by all means, submit a PR and as long as it's functional I'm 100% confident it will be merged into INAV.  But when you're asking other people to do it for you, and those people see no value in it, you should accept their reasoning not to.

Eventually, someone who wants DSHOT will submit a PR and it will be added.  So far, I haven't seen anyone come forward to actually do the work, only relentless demands from others not interested in lifting a finger, and not seeing other's perspective on why they should take time away from their family to program something for free that they will never use.

I hope that makes at least my perspective on this perfectly clear.

TwoToneEddy, 2018-08-02:
@teckel12 Well actually I did volunteer to do the work a couple of posts back in this issue so I fully intend on lifting my finger.

With regards to the whole digital Vs analogue thing, pwm, one shot etc rely on the escs interpretation of pulse width, D-shot doesn't. I'm absolutely not saying digital is better than analogue, it of course depends on the application. When it comes to communication however I believe the data being digital is superior. Ethernet isn't digital on the wire it uses an analogue carrier but the information is digital, it's the information interpretation that's important.

Digital headphones and speakers have a completely different purpose so that's irrelevant.

As I said, I'll implement it once I get my head around it's exact implementation and how it fits into betaflight.

V8Darren, 2018-08-05:
Coming from a pure user point of view, I'm really hopeful for BLHeli_32 and DShot integration. Forgetting about the analogue vs digital and speed vs reliability arguments, to me it's about usability.

1. The ease of ESC calibration isn't as quick and simple as @teckel12 suggests, especially in the field. It's not a case of arm, raise the throttle, stick in the battery, and lower the throttle after the beeps anymore. iNav adds the issue of needing GPS lock before you can arm, which means you need to do it with a laptop via the motors tab.

2. There are a lot of BLHeli_32 ESCs coming to market, and most new ESCs are DShot. The issue lies in that some of these ESCs can no longer be programmed through iNav. I have ESCs right now (Racerstar Tattoo+ 52A) which I cannot with BLHeli Configurator through an iNav flashed board. This ESC has no beep commands to set things like ESC breaking, timing, or reversing, so currently I'm stuffed unless I either pull the ESC and buy a different interface to configure it or flash the board with Betaflight, make my changes, and re-flash it with iNav. All of which is a complete pain and completely unnecessary.

The market is going BLHeli_32 and DShot, the only options to me are implement it, or provide a list of unsupported ESCs; both need work and maintenance, however the former is an inclusive strategy, the latter exclusive.

TwoToneEddy, 2018-08-05:
And when you crash in a field a mile away turtle mode will be quite handy ;-)

teckel12, 2018-08-05:
@V8Darren
1. Completely untrue. I've always calibrated my ESCs on the bench and have never used a laptop. I don't even have a laptop.
2. I've had zero problems finding ESCs that support DSHOT as well as Multishot and program just fine through INAV. I'm also don't let marketing hype get to me so I know 32 bit isn't better than 16 bit when it comes to ESCs.

V8Darren, 2018-08-05:
@teckel12 

1. Is completely true, you’re changing the parameters. Perhaps I was too specific saying laptop when computer should have been used, but I added the caveat of “especially in the field” where you’re hardly going to take a desktop. To calibrate the throttle using the transmitter alone, you’d either have to either preemptively or permanently disable safety checking so the craft could arm immediately without, for example, waiting for a GPS fix, which isn’t dangerous, or have some other way to power the flight controller in the field without the main battery, assuming the GPS is powered by the flight controller and not a PDB. I agree at the bench it’s easy enough to do, but you’re not always at the bench when you want to calibrate the ESC,for varied reasons. 

2. I’m not letting marketing do anything. I spec’d my ESC based on size, amp rating, weight, and cost, it just so happens it’s a BLheli_32 ESC with support for DShot 1200. It also just so happens that trying to program it via iNav doesn’t work (works fine through Betaflight, but that plays havoc with servos, so I’d rather not).

TwoToneEddy, 2018-08-05:
@teckel12 If you're building a racing style quad with GPS functionally ( with the GPS mounted in a decent place wrt protection) then turtle mode is worth having. I'm building just such a quad that flies like a racer / acro quad but can return to home if I need it.

32 bit is absolutely better than 8 bit. In the case of Blheli 32 the processor used is also faster allowing 48kHz pwm which can help control lower inductance (higher KV) motors better (more efficiently). Having the extra processing power for telemetry could also be handy for inav, for example, you could have the current and motor rpm read out on the OSD and the quad goes down, you have a means of postmortem. The industry is going to Blheli 32, like it or not.

teckel12, 2018-08-05:
@V8Darren You and I must calibrate ESCs differently. I do it only when I install a new ESC, which is like once a year. And then I need to be on the bench anyway to solder. And there nothing dangerous about it as the motors never spin and INAV isn't even involved.

Basically, this has never (even remotely) been an issue even in the slightest. 

And don't forget, you loose the ability to flash or reprogram your ESCs in the field with blheli_32 as it requires an internet connection. 

If blheli_32 was open source and didn't require an internet connection I'd be okay with it. But for those reasons it's a no-go for me. 8-bit works just fine for Multishot, and faster and more reliable than DSHOT. And there's plenty of non blheli_32 ESCs out there. I've seen no shortage.

TwoToneEddy, 2018-08-05:
@teckel12 Every person I've spoken to at the various clubs I fly at in the UK has complained that D-shot isn't supported.

digitalentity, 2018-08-05:
Guys, DSHOT for INAV is coming in post 2.0, I suggest we end arguing about it's usefullness, especially on a closed issue.

terragady, 2018-08-05:
@TwoToneEddy I do not think you can complain for lack of features in something which you get for free from other people. What they can do, instead of complaining, is implement it! Simple as that :)

digitalentity, 2018-08-05:
Also, for the record. BLHeli32 works perfectly with INAV 2.0:

TwoToneEddy, 2018-08-05:
@terragady Complain is probably the wrong word, they wonder why it isn't there. Granted they're not programmers so don't appreciate what's involved.

@digitalentity Agreed, let me know when you have the timer and I'll do my best to port the rest from betaflight. In the meantime I'll look at how it's done in betaflight so I'll hopefully be in a position to implement it when the time comes.

digitalentity, 2018-08-05:
@TwoToneEddy let's move the discussion about actually getting things working to https://github.com/iNavFlight/inav/issues/1393. I'm slowly working on timer stuff, might take a few weeks given limited amount of time available.

lightning-100, 2019-02-03:
I really cant believe this is a discussion, It is a must have for me. And yes i am aware it is free and most do not contribute like cleanflight. betaflight, ardupilot. I dont mean to be short or rude but  contributors dont see the use in implementing esc telemetry really!

teckel12, 2019-02-03:
@lightning-100 I really don't care at all, to each his own. But I'm sure someone out there will find this interesting enough to do it. The DSHOT protocol isn't even released yet, I think you're getting a bit ahead of yourself.

digitalentity, 2019-02-03:
@lightning-100 DSHOT telemetry is in the works for 2.2 - https://github.com/iNavFlight/inav/issues/2710. DSHOT is available for select boards in 2.1

CorruptName, 2020-03-31:
I know this stuff is old, but it's funny to see @teckel12 spent so much effort fighting against DSHOT.