iNavFlight/inav #1504: request fot differential thrust and torque roll.

Guycho, 2017-04-02:
Hi all, this is a request to include differential thrust for twin motor flying wings,
It would also be awesome to have torque roll through increasing power to one motor.

is that something that is possible to do?
on a traditional wing with no fc it works quite nice.

thank you,
Guy

Schalonsus, 2017-04-02:
Diff thrust for planes is already supported you just need custom motor mixing:
mmix reset
mmix 0 1.0 0.0 0.0 0.3  # Left Engine
mmix 1 1.0 0.0 0.0 -0.3  # Right Engine

digitalentity, 2017-04-03:
As @Schalonsus pointed out you already have a possibility to add differential thrust. What is torque roll, how should it function and how is it different from differential thrust?

Guycho, 2017-04-03:
i called it torque roll for lack of a better name (that i know of).
what i mean by it is using the motor to roll the plane by using the torque of the spinning motor (like how quadcopters yaw) newton's 3rd law:).

how it actually work is like this. you have two motors. the right one is cw and the left is ccw. when you want to roll left, not only the the elevons move but the right motor increases and the left motor decreases.
adding a bit of yaw and rolling the plane.
if diff thrust is allready supported , i can implement  torque roll by using a simple mix at the Tx.

thank you,
Guy

oleost, 2017-04-03:
@Guycho I would not do it that way.

I would assume you could simple make your own motor mix, with roll authority added to the motors aswell. 

Do some reading on https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md
And watch this https://www.youtube.com/watch?v=qL2LMFF5bJ4

This will probably give you an idea how to accomplish what you want. And then try and fail on bench :)

oleost, 2017-04-03:
I think somethink like this to give roll command to motors, you may have to play with invertion.

mmix reset
mmix 0 1.0 0.3 0.0 0.0 # Left Engine
mmix 1 1.0 -0.3 0.0 0.0 # Right Engine

or with both roll and yaw authority on motors.

mmix reset
mmix 0 1.0 0.3 0.0 0.3 # Left Engine
mmix 1 1.0 -0.3 0.0 -0.3 # Right Engine

Redshifft, 2017-04-03:
@Guycho You will likely introduce much more YAW than roll an Aeroplane has a full wing  counteracting/damping roll.

Guycho, 2017-04-03:
my idea is to work with the control surfaces, not by torque alone.
do you still believe it will not be enough?

thank you,
Guy

Redshifft, 2017-04-03:
Even an Aeroplane in a Hover with a single big propeller does not generate masses of roll, so a small propeller has little chance of contributing much.

Dronek, 2017-04-03:
> my idea is to work with the control surfaces, not by torque alone.
 do you still believe it will not be enough?

Maybe it would become clearer, if you defined the goal, you are trying to acheive with this approach.
What are the benefits of additional roll-authority by using torque for fixed-wing?
Does a problem for your solution actually exist?

The way I see it, a plane is constructed to be held in a stable orientation by airflow unless control surfaces are utilized. Changing motor speed rapidly in order to gain a lot of torque would probably do more harm to fixed-wing motor-mounts and respective parts of the airfoil due to increased mechanical stress. It would ideally require multirotor ESCs with active braking as well.
The only scenario where I see a benefit or "torque rolling" is some kind of active stabilization in stall situations.

digitalentity, 2017-04-04:
I agree with @Dronek. Torque roll will probably do more harm than good.

Guycho, 2017-04-04:
my goal was to have more authority at very low speed, not for stabilization, because it will yaw one way or the other.  i'll do some testing on my wings to see if it's actually worth it.

thank you all:)

Guy

digitalentity, 2017-04-04:
@Guycho please let us know about your findings.

DzikuVx, 2017-06-12:
Topic appears to be abandoned, so I'm closing it. If case anything new appears, feel free to reopen

closed 2017-06-12 by DzikuVx