iNavFlight/inav #1508: Fail Safe activate craft just falls out of the sky yet set to RTH

BoLtWD, 2017-04-03:
I am having a problem with Fail Safe using INAV. I have setup a switch to test Fail Safe but if I activate it the craft just falls out of the sky.

In INAV Fail Safe stage 2 is enabled, 
Fail Safe switch is enabled, 
Guard Time for stage 2 = 5, 
Fail Safe low throttle Low delay = 100, 
Land is set to on, 
Throttle value when landing = 1200, 
Delay for Turing off Motors = 200, 

XR4-SB RX has been setup for  no pulses, 
ALL AUX off inc motor start, Throttle off.

I have tested RTH and this works ok, but Fail Safe drops the craft out of the sky thou RTH has been selected, 

I know I am missing something can you help?

digitalentity, 2017-04-03:
Please provide a complete config dump

BoLtWD, 2017-04-03:
# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 397 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2017-04-03:
Any reason for turning on failsafe_kill_switch?

This basically makes your Failsafe mode act as an kill switch, bypassing everything failsafe related.

BoLtWD, 2017-04-03:
Yep I was trying this by turning it on to see if this would help but must still have it on. Yep just checked its on turned it off when outside powered up (Removed Props), got full GPS and craft was happy to fly, Armed the craft motors spinning and walked from Home point apx 80ft still with motors running then turned off the RC and the craft after 2 sec stopped all motors and beeped SOS. I did this again but this time evoked FS of a switch again is stopped all motors after 2 sec and beeped SOS. All this is with failsafe_kill_switch set to off.

oleost, 2017-04-03:
Share an dump where it fails and it's supposed to work....

BoLtWD, 2017-04-04:
Sorry for the delay in getting back to you..

I dont know if I have done it correct but this is what I did to get the dump..

1. Powered craft until GPS lock
2. Armed motors (Props off) then walked with craft apx 200ft
3. Powered off RC so no pulses.
4. Craft stops motors after 2sec and starts beeping (SOS)
5. Connected craft still powered and beeping to Laptop
6. CLI typed DUMP and got below

**Let me know if this is not what you needed 
and how I can provide with instructions.**
________________________________________________________________

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3 1.6.0 Feb 28 2017 / 13:44:32 (0def456)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
[... 416 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

oleost, 2017-04-04:
I can`t see anything wrong, make sure your RX sends no pulses, and not jumping to an random preset.
Also you could instead of using no pulse, jump to an failsafe mode. https://github.com/iNavFlight/inav/wiki/Failsafe

( Also regards to your dump, not necessary to do those steps. )

BoLtWD, 2017-04-04:
I will give these a go later and let you know

digitalentity, 2017-04-04:
I presume you don't raise throttle above zero after arming. This is expected behavior - if throttle is at zero for a while firmware will disarm instantly. Please increase throttle a bit (say 30%) before turning off the transmitter. If it still fails - we'll troubleshoot further.

BoLtWD, 2017-04-04:
Wow you gents think of everything, you were bang on the money and I was playing it safe but did not think of throttling up.
I armed the motors again after GPS fix then throttled up around 50% then walked 100ft, shut RC off and the craft beeped but kept the motors running so this must have been in RTH mode. 
I did the same but this time I set it to land, shut RC off the craft beeped and I placed it on the ground where after a time it powered off.

**Sorry for wasting your time digitalentity & oleost and thank you for all the hard work you do.**

closed 2017-04-04 by BoLtWD

digitalentity, 2017-04-04:
No worries @BoLtWD, good that you sorted it out.