iNavFlight/inav #1510: Set Altitude Ceiling

heathicus, 2017-04-03:
This is a feature request.  I'd like to be able to set an altitude ceiling that will prevent the craft from going about that altitude.  Similar to geofencing, but no lateral boundaries, only height.  The user would be able to set the altitude ceiling.  The software would check whatever sensors are available that report altitude.  If the reported altitude is equal to or greater than the set ceiling, the throttle would be reduced to either a value set by the user (which should be a value just under what is needed to hover) or to the minimum throttle value until the craft is below that ceiling.  

My use case for this is as a "trainer" mode for my kids.  I want to set a max altitude while they learn to fly FPV, but not trigger any RTH, or Failsafes, and keep all other controls responsive even when at or above the ceiling.  Only throttle should be lowered.

bzdang, 2017-04-03:
Great idea, I would love to have this feature.  Recreational flying in Canada is under the microscope, so I'd like to be able to demonstrate good flying practices to anyone who shows up where I fly. Some are curious and some are hostile to the activity, but each visit is a teaching opportunity. Using tech to voluntarily set a safe altitude limit shows them that I care about safety. 
Almost no-one in the general public knows that we're getting telemetry back from our aircraft - if they did, they'd realize just how criminally irresponsible are the 'train-spotters' who try to get close-ups of aircraft. These are never accidental events and the perpetrators should be hunted down like criminals, just like those who point lasers at aircraft.
Dave
Toronto Canada

Dronek, 2017-04-07:
I also like the idea of altitude ceiling, especially in the light of recently updated regulations for flying UAVs in Germany. We always try to encounter asking people with as much competence as possible and explaining such safety functions is very helpful when trying to diminish exaggerated "drone-fear".

However, rather than altitude ceiling only, it would be even nicer to have some kind of altitude band with upper and lower limit **as a mode**.
This way there would be more flexibility to define altitude ceiling only (altitude bottom set to -999999m), altitude bottom only (altitude ceiling set to 999999m) or altitude band (e.g. 30m to 100m) and switch it on or off during flight. Altitude band would certainly be even more safe for training purposes.

digitalentity, 2018-02-25:
Rewrite for failsafe system is in the works. Breaching the altitude could be one of failsafe triggers.

bzdang, 2018-02-26:
Triggering failsafe might cause undesired behaviour, depending on the configured failsafe procedure!
Better if it behaves as though ALTHOLD was switched on as the aircraft exceeds the ceiling altitude, but with only decending or holding at ceiling permitted until some distance below the ceiling. Thanks for listening and all the great work!