iNavFlight/inav #1514: Buzzer should be quiet when -ON_USB is set.

krzysztofmatula, 2017-04-04:
Making the buzzer quiet by setting "beeper -ON_USB" doesn't seem to work on my setup (diff below).

I usually connect (and power) the FC by USB for log downloading. Inav configurator displays MAG and GPS in red, also monitored voltage may be in alarming state. But that's fine. What is annoying, is that buzzer keeps producing sound, despite the fact I set "-ON-USB". I think the setting should be global, above any other buzzer related ones.

```
# diff

# version
# INAV/SPRACINGF3 1.6.1 Mar 28 2017 / 18:51:24 (a3c93f9)

# resources

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RSSI_ADC
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map

# serial
serial 0 1 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 2 16 57600 57600 57600 115200
serial 31 1 9600 38400 0 115200

# led
led 0 15,15:S:C:2
led 1 14,15:S:CW:2
led 2 9,15:S:CW:2
led 3 8,15:S:CW:2
led 4 7,15:S:CW:2
led 5 6,15:S:CW:2
led 6 1,15:S:CW:2
led 7 0,15:S:C:2
led 8 11,0:N:C:1
led 9 10,0:N:CW:1
led 10 9,0:N:CW:1
led 11 8,0:N:CW:1
led 12 7,0:N:CW:1
led 13 6,0:N:CW:1
led 14 5,0:N:CW:1
led 15 4,0:N:C:1

# color

# mode_color

# aux
aux 0 2 1 900 1200
aux 1 4 1 900 1075
aux 2 4 1 1250 1350
aux 3 3 0 1100 1250
aux 4 3 0 1350 1475
aux 5 3 0 1600 1750
aux 6 3 0 1900 2100
aux 7 9 0 1250 1500
aux 8 8 0 1500 2100
aux 9 11 1 1500 2100
aux 10 12 2 1700 2100
aux 11 18 1 1500 2100

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set acczero_x = 48
set acczero_y = -31
set acczero_z = 12
set accgain_x = 4082
set accgain_y = 4078
set accgain_z = 4078
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 116
set magzero_y = -54
set magzero_z = -17
set baro_hardware = MS5611
set blackbox_rate_denom = 4
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set nav_extra_arming_safety = OFF
set nav_mc_hover_thr = 1300

# profile
profile 1

set roll_rate = 40
set pitch_rate = 40

# 
```

digitalentity, 2017-04-04:
It's already fixed in latest development.

DzikuVx, 2017-05-23:
Yes, this is fixed, closing

closed 2017-05-23 by DzikuVx

lancaster100, 2018-01-16:
Hi DzikuVx
I am trying to understand how the CLI controls the choice of beeper sounds
I have enabled two sounds but my CLI dump lists 21 sounds - does this mean they are all enabled?
I notice in the above comments that a negative sign is placed ahead of the command line 
'beeper -ON_USB'  is this how beeper sounds are disabled?
regards
Hans

Think-Fly, 2018-02-26:
yes,'beeper -ON_USB' is how beeper sounds are disabled.The followings are copied from my INAV cli.
# beeper -ON_USB
Disabled ON_USB

krl91, 2023-05-07:
I have the same issue on iNav 6.0.0 ...