iNavFlight/inav #1525: iNav Config 1.6.3 - new preset 5" GPS - MAG ging crazy

Antonio76m, 2017-04-08:
hi all, thanks in advance,
I saw there is this new preset for 5" GPS, 
I have a 330 class with F3 Deluxe equipped with a UBlox GPS M8N
if I set 5" Racer MAG is ok but if I select the preset 5" GPS, on the SEUP main page the Heading shows always 0 and if I move the quad it does not move in picture,
is it safe to fly? does anybody knows how to fix? or any suggestion on setting to select the 5" Racer and optimize for GPS?
attached 2 screens to try to make sanse on what I'm saying :)

[image]
[image]

 Thanks Antonio

oleost, 2017-04-08:
Heading should move.

But what does the mag light show? And what magnetometer is actually selected and have you tried doing a new auto detect mag?

oleost, 2017-04-08:
Also what does the accelerometer sensor light give you?

Antonio76m, 2017-04-08:
what do you mean with "mag light" and "accelerometer sensor light" ?
are you referring to the attached?
yes I've tried MAG auto detect and it gives me the HMC5883
thanks

[image]

Antonio76m, 2017-04-08:
as I'm not expert, below the only difference I can see, I changed them but still problem persist
thanks
[image]
[image]

oleost, 2017-04-08:
Hmm, can confirm issues. Unsure why its like this. Guess we have to wait for someone else to come online.
It appears when enabling the notch filter.
```
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
```

This is raw diff dump from GPS preset. When this is applied GUI model is shaky, heading is not working. ( it actually shows heading change when pitching, instead of when yawing.

```
INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6050
set mag_hardware = NONE
set baro_hardware = MS5611
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
```

Removing the notch changes it behaves as expected:

```
INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)
set gyro_sync = ON
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 90
set acc_hardware = MPU6050
set mag_hardware = NONE
set baro_hardware = MS5611
set motor_pwm_rate = 1000
set motor_pwm_protocol = ONESHOT125
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
```

@DzikuVx Any idea?

DzikuVx, 2017-04-08:
hmm... I do not have SPRF3 to confirm, but:

1. I'm using GPS with SPRF and notches on my big quad, no problems
1. There is a chance that it is connected with overall system load. With notch enable, it's just too much

@oleost could you verify with `i2c_overclock = ON` ?

Antonio76m, 2017-04-08:
Hi guys, I've also a spare SP RACING F3 EVO, still to be soldered etc, I'v installed latest iNav as per F3 Deluxe and the problem still persist also in the other board "naked" no motors, GPS etc connected, just the board. thanks

andre92, 2017-04-08:
@Antonio76m SP Racing F3 deluxe has a HMC5883 built in. Many GPS boards come with that same sensor. Do you have a board without the HMC5883 to test the gps module's compass? It could be a conflict of 2 identical chips on your i2c bus. As for the notch filter, you should check cpu load, and task timing, maybe enable i2c_overclock (iNav defaults are safe, Betaflight removed option to have it off). Notch filters can only work if desired sampling rate is achieved, if not they will produce garbage.

Antonio76m, 2017-04-08:
@andre92 thanks for the reply, thing is my GPS has no MAG in it, so no conflict there :(
I did try set i2c_overclock=ON just now but it does not solve the issue

andre92, 2017-04-08:
Does it show any i2c errors? and could you post the output of the commands "status" and "tasks"?

Antonio76m, 2017-04-08:
how do I do that? sorry I'm not expert :(

andre92, 2017-04-08:
You can type those commands into the CLI. It should output the rate and duration of each task. That way we can determine whether the cpu is overloaded. 
Please paste the output for both commands, and put them into code tags. Paste, select and press this button: <>

Antonio76m, 2017-04-08:
`System Uptime: 12 seconds
Voltage: 11 * 0.1V (1S battery - CRITICAL)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=MS5611, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=NONE
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 3234, max available config: 4096
System load: 4, cycle time: 2018, PID rate: 495, RX rate: 49, System rate: 9`

`acc_task_frequency = 500
Allowed range: 100 - 1000

attitude_task_frequency = 250
Allowed range: 100 - 1000`

Antonio76m, 2017-04-08:
is this ok?

andre92, 2017-04-08:
Status looks ok to me, at least if "system load" is in percent, that would mean 4% load and maybe you could run it at 1khz too. Could you post the output of the "tasks" command as well?

Antonio76m, 2017-04-08:
`Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9       5       3    0.5%    0.5%         2
 1 -          PID     495     717     689   35.9%   34.6%     22274
 5 -           RX      49     116     107    1.0%    1.0%       366
 6 -       SERIAL      99     134       5    1.8%    0.5%       156
 7 -      BATTERY      49   40185       6  197.4%    0.5%        62
 8 -       BEEPER      99      23      11    0.7%    0.6%        72
 9 -          GPS      24     129      19    0.8%    0.5%        25
10 -      COMPASS       9     236     233    0.7%    0.7%       154
11 -         BARO      99     246     240    2.9%    2.8%      1588
15 -     LEDSTRIP      99     406     248    4.5%    2.9%      1800
17 -          CMS      59       8       5    0.5%    0.5%        18
Task check function             7       6                        17
Total (excluding SERIAL)                   244.9%   44.6%
`

andre92, 2017-04-08:
let me clean that up a bit and sum it up
`Task list      rate/hz   max/us  avg/us  maxload avgload total/ms 
0 - SYSTEM      9       5           3          0.5%       0.5%      2 
1 - PID              495   717       689       35.9%    34.6%     22274 
5 - RX               49     116       107      1.0%       1.0%       366 
6 - SERIAL        99    134        5          1.8%       0.5%      156 
7 - BATTERY    49     40185   6          197.4%   0.5%       62 
8 - BEEPER      99     23         11        0.7%       0.6%       72 
9 - GPS             24    129       19         0.8%      0.5%       25 
10 - COMPASS  9     236        233      0.7%       0.7%      154 
11 - BARO         99    246        240      2.9%      2.8%       1588 
15 - LEDSTRIP 99     406        248     4.5%       2.9%       1800 
17 - CMS          59      8           5          0.5%       0.5% 
18 Task check function 7 6 17 
Total (excluding SERIAL) 244.9% 44.6%`

Antonio76m, 2017-04-08:
btw I've tested all the preset, just in case, and the issue happen only with 5" GPS, all other preset looks good.

digitalentity, 2017-04-09:
@oleost @Antonio76m what is your `gyro_rate_denom` set to?

oleost, 2017-04-09:
@digitalentity This preset was applied on a clean board, so `gyro_sync_denom` = 2

Tried changing to gyro_sync_denom = 4, did not help.

digitalentity, 2017-04-09:
@oleost this is the problem. Default preset samples gyro at 500Hz, meaning only frequencies up to 250Hz (Nyquist frequency) can be detected, sampled and filtered. However our notch filter is set to 400Hz center frequency which is waaay above the Nyquist frequency. We need to either disable notch or set it to something that is compatible with 500Hz sampling rate.

oleost, 2017-04-09:
Unsure what would give the best result for an 5inch racer?

Any qualified guess?

digitalentity, 2017-04-09:
I believe it should be enough to set to this:
```
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set gyro_notch2_hz = 0
set gyro_notch2_cutoff = 1
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
```

digitalentity, 2017-04-09:
I'm also adding a safeguard against this which will prevent filters from initializing with impossible settings.

closed 2017-04-09 by digitalentity

andre92, 2017-04-09:
How about setting gyro sampling at 1k? that will allow notch at 400hz. I managed that even on a F1.

digitalentity, 2017-04-09:
Won't be possible with althold or poshold enabled on F1. Even F3 will have some hard time dealing with the load.

Antonio76m, 2017-04-09:
@digitalentity @andre92 hi thanks fo the above, but I'm lost :)
how do I fix this? I try to set the above command but I still se the issue

Antonio76m, 2017-04-09:
sorry, I removed my last comment :) I didn't type SAVE hahaha

thanks

DzikuVx, 2017-04-09:
@andre92 ATM we can not set `i2c_overclock` over MSP
Perhaps it's time to change that

andre92, 2017-04-09:
> ATM we can not set i2c_overclock over MSP
> Perhaps it's time to change that
It is a CLI only setting. Betaflight doesn't have the option, but they have it always overclocked

andre92, 2017-04-09:
> ATM we can not set i2c_overclock over MSP
> Perhaps it's time to change that

It is a CLI only setting. Maybe put it next to looptime settings, and maybe include a warning that some hardware may not like it. Betaflight has it overclocked with no way to turn it off but I never heard of issues with it.