iNavFlight/inav #1526: Heading not restored on position hold after external perturbance

giacomo892, 2017-04-08:
I enabled position hold on my F450 (+altitude hold) then I picked a leg with an hand and forced a rotation in mid air more than one time. 

The quad started to respond but the final heading is not the one it was before I pushed the quad.

Shouldn't the heading be restored to the original one having the compass as reference? (MAG_HOLD?)

Blackbox log is avail if needed.

```
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
serial 0 128 115200 38400 57600 250000
serial 1 2 57600 115200 0 115200
serial 30 17 19200 38400 19200 115200
aux 0 1 0 950 2050
aux 1 1 1 950 2050
aux 2 3 0 1600 2025
aux 3 3 1 1600 2025
aux 4 9 1 1600 2025
aux 5 8 3 1700 2050
aux 6 19 2 1850 2050
set gyro_hardware_lpf = 98HZ
set gyro_lpf_hz = 42
set acc_hardware = MPU6050
set acczero_x = 164
set acczero_y = 16
set acczero_z = -12
set accgain_x = 4071
set accgain_y = 4082
set accgain_z = 4012
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set mag_declination = 230
set magzero_x = 55
set magzero_y = -133
set magzero_z = -33
set mag_hold_rate_limit = 45
set baro_hardware = BMP280
set rx_min_usec = 989
set rx_max_usec = 2100
set blackbox_rate_denom = 2
set motor_pwm_protocol = ONESHOT125
set failsafe_procedure = RTH
set align_board_roll = 4
set align_board_pitch = -8
set vbat_scale = 50
set gps_sbas_mode = EGNOS
set inav_use_gps_velned = NONE
set inav_reset_altitude = EACH_ARM
set nav_user_control_mode = CRUISE
set nav_max_speed = 600
set nav_max_climb_rate = 400
set nav_manual_speed = 1400
set nav_manual_climb_rate = 300
set nav_mc_hover_thr = 1510
profile 1

set mc_p_pitch = 64
set mc_d_pitch = 45
set mc_p_roll = 64
set mc_d_roll = 45
set mc_i_yaw = 40
set max_angle_inclination_rll = 350
set max_angle_inclination_pit = 350
set dterm_lpf_hz = 20
set rate_accel_limit_roll_pitch = 6000
set rate_accel_limit_yaw = 1800
set roll_rate = 35
set pitch_rate = 35
```

digitalentity, 2017-04-09:
Enable MAG mode.

giacomo892, 2017-04-09:
Will try and report. Sorry for the hassle

giacomo892, 2017-04-09:
Ok. Enabled MAG and all works as you said.
Only fact it is a bit bumpy when restoring the heading but it is expected since MAG has only a P part and the rotation I impose with hands is brutal respect to the normal ones like during a descent.
The 'issue' is solved.

closed 2017-04-21 by digitalentity