iNavFlight/inav #1528: Altitude Hold with baro only in airplane

Hwurzburg, 2017-04-08:
nav_althold does not seem to work at all without a gps connected....I have tried the defaults as well as setting the nav_fw_z_p and _i values to high (100+) values and setting gps weight to 0 and baro weight to 1.0...while the throttle is set at cruise value and AH hold is indicated as being active in the osd, there is no variation in throttle or pitch as altitude freely wanders in turns or over short/long term...the plane is trimmed well and slowly varies alt in angle only mode....so its not a matter of being unable to overcome some kind of radical miss-trim...

it works well and is observably tuneable by those params, with gps+baro in another model...

digitalentity, 2017-04-09:
Blackbox logs and configuration dump would definitely help.

Hwurzburg, 2017-04-09:
no backbox log available (cant add an sd card without ripping the fc out of the plane...if I get time next week, I might try to add one)...but here is the diff..
[z-84.txt](https://github.com/iNavFlight/inav/files/908182/z-84.txt)

Hwurzburg, 2017-04-14:
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/922556/LOG00001.TXT)
here is a log file...I had the logs run any time ANGLE mode was on...and it has several segments with ANGLE+AltHOLD during a short flight

digitalentity, 2017-04-16:
@Hwurzburg thanks for logs. Bug found. It was a stupid typo in the code that was checking for **position** sensor (GPS) instead of **altitude** sensor when figuring out if altitude hold is possible.

digitalentity, 2017-04-16:
Will be fixed as part of https://github.com/iNavFlight/inav/pull/1542

DzikuVx, 2017-06-09:
Since fix was already merged, I'm closing this one. Thank you

closed 2017-06-09 by DzikuVx