iNavFlight/inav #1530: Dodo Smart port Telemetry

Dutch-Drone-Dude, 2017-04-08:
I want to report that the smart port  telemetry isn't working with the dodo.
I have test it with a X4R-SB and a dedicated part connection and it wasn't working.
I did changed the X4R-SB for a brand new one and that wasn't the solution. I did flashed the board with cleanflight and betaflight and both worked directly without changing anything on the hardware.
inav version 1.6.1
Status dump:
`# status
System Uptime: 3866 seconds
Voltage: 0 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050.n, BARO=BMP280, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, SONAR=NONE, GPS=OK
Stack size: 2048, Stack address: 0x10002000
I2C Errors: 0, config size: 2514, max available config: 4096
System load: 4, cycle time: 2010, PID rate: 497, RX rate: 49, System rate: 9`

Bootlog dump:
`# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (0, 0, 0, 0)
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (11, 11, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (12, 12, 0, 2)
 502:  2          PWM_INIT_DONE 
 548:  9         GYRO_DETECTION  (2, 0, 0, 0)
 664: 10          ACC_DETECTION  (3, 0, 0, 0)
 685: 11         BARO_DETECTION  (4, 0, 0, 0)
 685: 12          MAG_DETECTION  (2, 0, 0, 0)
2348: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2348:  4       SENSOR_INIT_DONE 
2851:  5          GPS_INIT_DONE 
2851:  7    TELEMETRY_INIT_DONE 
2901:  8           SYSTEM_READY `

Dump dump:
`
# version
# INAV/RMDO 1.6.1 Mar  9 2017 / 00:46:03 (6c878ae)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
[... 435 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500`

turbosi92, 2017-06-12:
I'm seeing this same issue.  Works fine on betaflight but not on iNav.  Is there a fix?

flyfisher2, 2017-06-12:
The RMDO hex is for an old version of the board with limited memory. SPORT telemetry is not supported in this version.
Use the SPRACINGF3 hex instead. The hardware is the same, almost.

DzikuVx, 2017-08-28:
I think we can close it now

closed 2017-08-28 by DzikuVx