iNavFlight/inav #1533: toilet bowling and fly away

killerm1, 2017-04-09:
Hi,
im using omnibusf4sd and having a problem with PH. It flies fine in all modes except PH and RTH.
In PH it starts to toilet bowl and in RTH it flies away. Im assuming this must be a compass issue and have tried re calibrating but no difference. Any ideas?

Also when testing the compass and connected to configurator the heading seems to work fine when flat and level but if i change the pitch or roll the heading changes approximately equivalent to the pitch or roll angle. For example at a heading of 0 when i pitch up 10 the heading changes approx 10. Is that normal?

giacomo892, 2017-04-09:
Are you using an internal compass? Is so use an external one.

killerm1, 2017-04-09:
Sorry using external compass. There is no internal on the omnibus fc.

giacomo892, 2017-04-09:
Maybe it is too close to the power lines.

giacomo892, 2017-04-09:
And also.. if the heading change when you pitch and roll please fix mag orientation from the GUI.

Xe54, 2017-04-09:
Sounds like a incorrect setting for your mag alignment to me. Check which way the mag is mounted on the board an set the correct mag alignment in configuration tab.

stagesixcup, 2017-04-10:
i noticed this "Issue" also, running my ZMR250 Quad with good Acro PID setup, Level worked fine, a.s.o., My Setup with a M8N Mini GPS and with GPS_AUTO_CONFIG=ON+AUTO_BAUD=ON doesnt worked well (Flyaways+Toilet Boiling). 

I decided to setting it up in U-Center with the Manual Settings from INAV Site (Baud=57600) and then it worked very well with 13-14 SAT's outside. BUT I have at PH wobbles and oscillations on both axis, i actually dont know whats wrong, i tried at POS XY many PID settings...

krzysztofmatula, 2017-04-10:
From the symptoms I'd suggest to:
- double check the mag orientation setting
- perform calibration (even if done before, what I observed - after software upgrade the calibration values may be different)

Then before actually trying to fly check the compass just on the desk:
- the heading must be correct and stable when rotating the model
- the heading SHALL NOT change when tilting the model in pitch and/or roll axis.

When the above is correct and it still does toilet-bowling on PH, then the compass is probably too close to power lines.

alfag4, 2017-04-10:
I have the same problem but only with the RTH waypoint during the landing phase. If enabling RTH from a switch my quad return and land without bowling.

andre92, 2017-04-10:
support issue

digitalentity, 2017-04-12:
This is definitely a setup issue. Check your mag alignment and verify heading as recommended by @krzysztofmatula

closed 2017-04-12 by digitalentity

peakhunt, 2017-05-03:
@stagesixcup
interesting since I am having the same toilet bowling issue with my 280 quad.
let me set up my ublox-m8 manually as you suggested and test it.

Atorhe, 2017-09-17:
Hi! I am having the exact same issue on my omnibus! Did you ever solve this problem?

digitalentity, 2017-09-18:
Toilet-bowling is always a compass issue. Please make sure you compass is giving valid readings and is not affected by magnetic fields.

giacomo892, 2017-09-18:
I had a toilet bowling issue some time ago and it was due a failure in compass calibration value restore after upgrade. 
Somehow on of the last digit of the three parameters went missing while sending to the FC.
I think that's because CLI doesn't have any CRC.

Atorhe, 2017-09-27:
How do i check if i have valide compass readings in inav?
I tried to elevate the gps unit another 5cm to see if that made any difference, but it stays the same... now the compass is about 12 cm avay from any other electronics.

stronnag, 2017-09-28:
See the wiki. A blackbox log of flight in straight lines at reasonable speed so the mag and gps readings can be correlated.

swemonstro, 2018-03-26:
Picking up on this thread since I have the same problem. I have confirmed that my compassheading is correct by flying in a straight line as suggested and the mag and the gps direction is basicly the same.

Whats the next step, since the toilet bowl is still there?

han8jun, 2018-04-28:
Is there anyone has succeeded POSTION-HOLD with Internal Compass of FC ?  Abservably a external compass could fix it, but I just want to know that is it neccessry to do extra try.

DzikuVx, 2018-04-28:
@han8jun maybe someone did. But doubt it. For the rest of users (99,99%) external is just required