iNavFlight/inav #1535: Enabling of HMC5883 impacts functionality of my gimbal on SPRF3

Xe54, 2017-04-09:
I discovered a weird dependency between connecting/enabling of an external mag (HMC5883) and functionality of my gimbal on SPRacingF3 (Acro) board. The gimbal works fine if the mag is either disconnected or disabled in the software (MAG_HARDWARE = NONE). After enabling the mag I can not control my gimbal from RC. I've testet this behavior on two different M8N modules with HMC5883 with the same result. The gimbal is connected to the pins 7 and 8 of the fc. Here are my bootlog and dump:

# bootlog
Time Evt            Description  Parameters
   0:  0          CONFIG_LOADED 
   0:  1       SYSTEM_INIT_DONE 
 502: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 502: 19   TIMER_CHANNEL_MAPPED  (7, 6, 1, 3)
 502: 19   TIMER_CHANNEL_MAPPED  (8, 6, 2, 3)
 502: 18  TIMER_CHANNEL_SKIPPED  (9, 6, 2, 3)
 502: 18  TIMER_CHANNEL_SKIPPED  (10, 6, 2, 3)
 502: 18  TIMER_CHANNEL_SKIPPED  (11, 6, 2, 3)
 502: 18  TIMER_CHANNEL_SKIPPED  (12, 6, 2, 3)
 502:  2          PWM_INIT_DONE 
 512:  3       EXTRA_BOOT_DELAY 
1050:  9         GYRO_DETECTION  (2, 0, 0, 0)
1166: 10          ACC_DETECTION  (3, 0, 0, 0)
1180: 11         BARO_DETECTION  (3, 0, 0, 0)
1180: 12          MAG_DETECTION  (2, 0, 0, 0)
2845: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2845:  4       SENSOR_INIT_DONE 
3398:  5          GPS_INIT_DONE 
3449:  7    TELEMETRY_INIT_DONE 
3500:  8           SYSTEM_READY 

# 

# dump

# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 5
servo 1 1000 2000 1500 90 90 100 6
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
[... 433 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 18
set pitch_rate = 18
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

hali9, 2017-04-14:
I use

Xe54, 2017-04-17:
@hali9 Baro is definitely not the problem. My set up works fine except of the gimbal. But I think I figured out what it could be. My guess is, that after enabling of the HMC5883 the board produces some signals on pins 7 and 8 while starting up. Those signals put my gimbal in an adjustment mode or something like that. If I switch my gimbal off and on while board is still running, it becomes controllable again. I can not proove it, but it must be some signals at start-up.

digitalentity, 2017-04-17:
The board might produce garbage on the outputs until it's fully initialized. Compass detection makes initialization longer - this might be the problem.

Xe54, 2017-04-17:
Is there a way to solve it?

digitalentity, 2017-04-17:
No, there will be a low level pulse on the outputs until the board enables outputs and initializes - not much we can do about it.

Xe54, 2017-04-18:
Ok, I will close this one. Is there a possibility to speed up the initializing process in the future?

closed 2017-04-18 by Xe54