iNavFlight/inav #1541: LEDLOW Not making sense

linuxras, 2017-04-09:
I LEDs have on my quad and need to be able to turn them off for all sorts of reasons.
I have assigned LEDLOW to a switch on my radio and to my surprise all that happens when I switch to LEDLOW it that the FC seems to stop sending updates to the LEDs which just freezes them in the current state.
I find this to be completely useless because the DOCS say Turns Off LEDS.

What should happen is turn all the leds to Black then stop sending the updates to them they are then effectively OFF.

krzysztofmatula, 2017-04-10:
As I use this feature I can confirm, that on 1.6.1 it is working as expected, making the LEDs to turn off  (tested on SPRACINGF3 and SPRACINGF3EVO).
Which platform and which software are you using?

linuxras, 2017-04-12:
@krzysztofmatula It turned out to be a bad connection on the signal wire on one of the LEDs and for some reason it played up just when I was changing switch state so it never gets the turn off signal, so would stay in their last state.
Sorry for the fuss

closed 2017-04-12 by linuxras