iNavFlight/inav #1549: Feature request: save and accumulate statistics (semi-)permanently on FC

Dronek, 2017-04-12:
In mwosd it is quite common to display flight statistics like traveled distance, max. distance from home, max. height, top speed, flight time etc. at the end of a flight. The data is usually lost when we disconnect the battery.

It might be quite interesting to have such data being saved **and added up** locally on the FC itself in order to keep track of absolute traveled distance, absolute flight time, absolute number of flights and optionally some absolute max. values. 

These values could be a subset of CLI parameters that are updated in defined time intervals while the aircraft is armed or at least on each disarm. It would allow the user to save and restore the data prior and after FC firmware updates, respectively.

maybeyes, 2017-05-04:
it would be just great! We need this feature )

krzysztofmatula, 2017-05-05:
I like this idea.
It seems to be enough to update flash on each disarm only.

Dronek, 2017-09-04:
Functionality available in current release. Therefore closing.

closed 2017-10-06 by DzikuVx