iNavFlight/inav #1550: NMEA GPS not working with Naze / Flip32

deluded, 2017-04-12:
I have a NMEA GPS connected to UART2 at 9600 on a Flip32 board. Current Cleanflight release picks GPS up correctly and shows pos + satellites. Same settings on iNav show no GPS data on the GPS tab.

I've checked the signal on a scope and it looks clean, I can see the GPS has a lock and is outputting valid NMEA.

I've given the code a quick look and couldn't see anything obvious - will dig a bit further when I have time...

digitalentity, 2017-04-12:
F1 boards support only UBLOX protocol due to flash size limitations.

deluded, 2017-04-12:
Thanks - that would explain a lot!

I will attempt a custom build to re-include it and drop something else, and if that fails, get a newer board.

closed 2017-04-12 by deluded

deluded, 2017-04-12:
Commented out blackbox and uBlox from common.h, added GPS_PROTO_NMEA and it now works!