iNavFlight/inav #1553: ANYFCF7 : i2c mag and baro issues

urlu75, 2017-04-12:
Got the opportunity to try the BG [F7 board](http://www.banggood.com/STM32F745-100lqfp-216MHz-MPU6000-SPI-F7-Flight-Controller-for-FPV-Racing-Support-Betaflight-p-1137386.html).
FW 3.1.6, target=ANYFCF7.
External mag is detected (HMC5883L) but it's not working :

```
# status
System Uptime: 30 seconds
Voltage: 0 * 0.1V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000
Sensor status: GYRO=OK, ACC=OK, MAG=FAILING, BARO=NONE, SONAR=NONE, GPS=NONE
SD card: Startup failed
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 1, config size: 3277, max available config: 262144
System load: 0, cycle time: 2006, PID rate: 498, RX rate: 49, System rate: 9

# bootlog
Time Evt            Description  Parameters
   5:  0          CONFIG_LOADED 
   5:  1       SYSTEM_INIT_DONE 
 506: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (11, 11, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (12, 11, 1, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (13, 11, 2, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (14, 11, 3, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (15, 11, 4, 3)
 506:  2          PWM_INIT_DONE 
 513:  3       EXTRA_BOOT_DELAY 
1200:  9         GYRO_DETECTION  (6, 0, 0, 0)
1501: 10          ACC_DETECTION  (7, 0, 0, 0)
1501: 11         BARO_DETECTION  (0, 0, 0, 0)
1501: 20        PITOT_DETECTION  (0, 0, 0, 0)
1501: 12          MAG_DETECTION  (2, 0, 0, 0)
2355: 15 HMC5883L_READ_OK_COUNT  (0, 0)
2355: 16   HMC5883L_READ_FAILED  (5, 5)
2355: 17    HMC5883L_SATURATION  (5, 5)
2455: 14        MAG_INIT_FAILED 
2455:  4       SENSOR_INIT_DONE 
4015:  8           SYSTEM_READY 

```

External MS5611 baro is not detected at all.
```

# status
System Uptime: 8 seconds
Voltage: 0 * 0.1V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=UNAVAILABLE, SONAR=NONE, GPS=NONE
SD card: Startup failed
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 2, config size: 3277, max available config: 262144
System load: 0, cycle time: 2007, PID rate: 498, RX rate: 49, System rate: 9

# bootlog
Time Evt            Description  Parameters
   5:  0          CONFIG_LOADED 
   5:  1       SYSTEM_INIT_DONE 
 506: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (7, 7, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (8, 8, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (9, 9, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (10, 10, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (11, 11, 0, 2)
 506: 19   TIMER_CHANNEL_MAPPED  (12, 11, 1, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (13, 11, 2, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (14, 11, 3, 3)
 506: 19   TIMER_CHANNEL_MAPPED  (15, 11, 4, 3)
 506:  2          PWM_INIT_DONE 
 506:  3       EXTRA_BOOT_DELAY 
1193:  9         GYRO_DETECTION  (6, 0, 0, 0)
1495: 10          ACC_DETECTION  (7, 0, 0, 0)
1505: 11         BARO_DETECTION  (0, 0, 0, 0)
1505: 20        PITOT_DETECTION  (0, 0, 0, 0)
1505: 12          MAG_DETECTION  (0, 0, 0, 0)
1505:  4       SENSOR_INIT_DONE 
3064:  8           SYSTEM_READY 

```

digitalentity, 2017-04-13:
@sambas any ideas?

sambas, 2017-04-13:
MAG should work, mapped to external i2c and enabled. Baro is mapped to internal i2c.

urlu75, 2017-04-13:
tried with an other mag. Same result and mag is perfectly working on some other boards.

urlu75, 2017-04-13:
Also tried to redefined baro device, just to see.
#define BARO_I2C_INSTANCE I2C_DEVICE_EXT
it's not working, can't connect to the board when baro is connected.

sambas, 2017-04-13:
I can check latest code with HMC mag on proto board later this week.

sambas, 2017-04-22:
MAG worked with current development branch and BG board. Make sure your connections are correct and you have powered the board via servo connector. USB doesn't power external devices except 3.3V pin on U4/5 connector.

`# version
# INAV/ANYFCF7 1.7.0 Apr 22 2017 / 20:08:04 (6b44e31)

# status
System Uptime: 31 seconds
Voltage: 0 * 0.1V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000, MAG=HMC5883
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=NONE, SONAR=NONE, GPS=OK
SD card: None inserted
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 1, config size: 3271, max available config: 262144
System load: 0, cycle time: 2009, PID rate: 497, RX rate: 49, System rate: 9

# `

closed 2017-04-22 by sambas