iNavFlight/inav #1557: Extrem Problems with high or full throttle nd Tricopter /with video

boyi01, 2017-04-14:
Hey guys,

i hope this is the right place for my Problem, if not pls tell me ;)

I use i nav, at the moment without gps, because it's brocken ... and I'm a beginner... but today i found a bigger Problem... when do full or high throttel, my Tricopter is crashing... like in the Video

[https://www.youtube.com/watch?v=SL7Vqdt-cB4](url)

I hope there is somebody who can help me!

Thanks a lot!

Here i have added my #dump

`# version
# INAV/SPRACINGF3 1.6.1 Mar 9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer TRI

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_PPM
feature VBAT
feature RSSI_ADC
feature PWM_OUTPUT_ENABLE

# beeper
[... 404 lines cut ...]
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500
`

digitalentity, 2017-04-15:
Looks like ESC desync. Do you have signal ground between ESC and FC connected?

boyi01, 2017-04-16:
Yeah i have... I will check the calibration and the protocol for esc, thanks for hint... But the desync musst be smnall, cause i can flay, with normal speed very long, also full speed forward or so is no problem... the only problem is when i want use full throttel ..

boyi01, 2017-04-17:
So i got checked my esc.. there hat linking Problem and the max throttel of one was incorrect... 
that fixed the roll, but still i get strong oscillation on the servo with hight throttel

digitalentity, 2017-04-18:
Now it's likely filtering, tuning or vibration issues. Hard do diagnose without blackbox logs.

boyi01, 2017-04-19:
Nothing eayser then this!
Her pls 
[error.19.4.2017.TXT](https://github.com/iNavFlight/inav/files/937530/error.19.4.2017.TXT)

DzikuVx, 2017-10-06:
Better late than never... I took a look at the log and I'd say that from software point of view everything looks good. It seems to be connected with vibrations or frame flex. Closing. If anything new appears, please reopen

closed 2017-10-06 by DzikuVx