iNavFlight/inav #1558: MSP_RX: throttle does not return to zero in case of "lost signal"

rodizio1, 2017-04-15:
Did some tests with MSP_RX in inav 1.6.1, it seems, the throttle channel is not returning to zero in case of "lost signal".

Steps to reproduce:
- Move channels 1-4 to "full-throw" (so that they show "2000" in Inav configurator receiver tab)
- Stop sending MSP_RAW_RC messages

Roll/Pitch/Yaw returns to center (1500), throttle stays at 2000.

CLI diff:

```
# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer FLYING_WING

# servo

# servo mix

# feature
feature -RX_PPM
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature RX_MSP
feature PWM_OUTPUT_ENABLE

# beeper

# map

# serial
serial 1 2 115200 38400 0 115200
serial 2 1 19200 38400 0 115200
serial 30 16 115200 38400 19200 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# master
set acc_hardware = MPU6050
set acczero_x = 48
set acczero_y = -39
set acczero_z = -52
set accgain_x = 4090
set accgain_y = 4109
set accgain_z = 4066
set mag_hardware = HMC5883
set magzero_x = -27
set magzero_y = -823
set magzero_z = 550
set baro_hardware = MS5611
set min_throttle = 1160
set failsafe_procedure = RTH
set align_board_roll = -12
set align_board_yaw = 2700
set gps_sbas_mode = EGNOS

# profile
profile 3
```

tonyintn, 2017-04-17:
Did you properly setup your RX? I would not rely on INAV for that you should setup up failsafe on your receiver properly using a no pulse failsafe or a user-defined failsafe.

digitalentity, 2017-04-18:
Please read release notes for 1.6 release here https://github.com/iNavFlight/inav/releases/tag/1.6 - especially the "failsafe" section.

Throttle behavior in 1.6.x is defined by `rx_nosignal_throttle` parameter.

rodizio1, 2017-04-18:
> Did you properly setup your RX? I would not rely on INAV for that you should setup up failsafe on your receiver properly using a no pulse failsafe or a user-defined failsafe method if you RX supports it, I find it safer that way.

The receiver simply stops sending MSP packets, i.e. "no pulse failsafe". More safe than programming the R/C receiver to pre-defined channel values, otherwise R/C receiver failures or wiring failures won't be detected.

> Throttle behavior in 1.6.x is defined by rx_nosignal_throttle parameter.

Checked that parameter, it's set to "HOLD". Setting it to DROP doesn't change that behaviour. Please note, the throttle _never_ returns to zero, left it like that for minutes. Also tried setting failsafe_procedure to "drop" in the GUI, same behaviour.

Wanted to try and see if motor behaviour is different than what the GUI throttle channel shows in the receiver tab, but somehow it won't arm using MSP_RX and the Aux1 channel. When using the R/C control sliders in the Inav configurator GUI this works as expected.

vinicentus, 2017-05-03:
It seems iNav is doing something strange there. A temporary solution might be to make your receiver failsafe set a different channel to high and configure iNav to RTH  (If you have GPS), or land when that happens.

digitalentity, 2017-10-29:
Failsafe was reworked since this issue. I believe it's no longer relevant

closed 2017-10-29 by digitalentity