iNavFlight/inav #1563: Request to add distance from home to OSD

TwoToneEddy, 2017-04-16:
I think this would be a really great addition and probably fairly trivial to implement. Any chance of adding it? Direction to home would also be great.

ABLomas, 2017-04-18:
This is already available, just configure (turn on/off required elements).
Proof - random flight video, see both elements in top-right corner: https://www.youtube.com/watch?v=JVu4Zl1ciFs

TwoToneEddy, 2017-04-18:
But there is no indication that this is inav or betaflight osd. This could be a minim osd. I've just checked and there are no available elements that give this functionality. I'd be keen to know the exact elements you refer to. Thanks

ABLomas, 2017-04-18:
I really don't get your issue. Did you tried, at least, to turn on "Direction to home" or "Distance to home" elements?
[image]

PaulHam211, 2017-04-21:
@TwoToneEddy Perhaps you were on an older version of the configurator? I tried on 1.6.1 and the new OSD options were not there. I downloaded 1.7.0 and they all show now.

Q for devs - Direction to home is a number currently, could this possibly be an arrow, pointing to the home pos?

digitalentity, 2017-04-23:
@PaulHam211 did you upload the font?

PaulHam211, 2017-04-23:
@digitalentity nope. ill give it a go. cheers

TwoToneEddy, 2017-04-23:
Sorry, my original issue was i was using an out of date version of the firmware. Those additional elements are there now. Sorry for the stupid mistake.

digitalentity, 2017-04-24:
@TwoToneEddy happens :smile:

TwoToneEddy, 2017-04-24:
Do I need to close this issue now?

closed 2017-04-25 by digitalentity