iNavFlight/inav #1564: Flying wing. motor will not ARM!

johnmh96, 2017-04-17:
i have built a ft spear wing and added an F3 flight controller. everything is working fine, angle mode acro mode etc. and the motor spin up when i test it in the motors tab. but when i try to arm the plane nothing happens. i have set an arm button that should work. but it does not. what can be the problem?? please help

digitalentity, 2017-04-18:
Did you calibrate your sensors? See https://github.com/iNavFlight/inav/wiki/Sensor-calibration
Do you have a GPS lock?
Is your aircraft almost level when you arm?
Are all the pre-flight check marks on the main Configurator screen show green ticks?

patrickschaerer, 2017-05-03:
I have the same problem. Configurator shows green ticks. But as soon as i arm nav changes to red and the motor doesn't spin.

digitalentity, 2017-05-04:
@patrickschaerer dump?

patrickschaerer, 2017-05-04:
[patrickschaererDump.txt](https://github.com/iNavFlight/inav/files/975535/patrickschaererDump.txt)

Does nav work without compass? I deactivated mine because I wanted to use the Serialport for Bluetooth communication.

DzikuVx, 2017-06-09:
I can confirm fixed wings arming properly in INAV 1.7.1
This has to be setup or hardware related. I suspect one of sensors failing . If any new info appears, please feel free to reopen or create new ticket

closed 2017-06-09 by DzikuVx