iNavFlight/inav #1566: Rising D-value at min throttle

Xe54, 2017-04-18:
Hi guys,

I noticed that the motors are constantly speeding up while the quadcopter (QAV540 frame) is staying on the ground. The result is that the quadcopter tries to take off at min throttle after a while. Blackbox says that D-value was constantly rising. Is this the normal behaviour, or is this the indication for a too high D-gain? Do I missing something? This happens even on the pretty plain ground. I must say, I've never noticed this behaviour with cleanflight on my setup. Here is the dump:

# dump

# version
# INAV/SPRACINGF3 1.6.1 Mar  9 2017 / 00:47:57 (6c878ae)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 5
servo 1 1000 2000 1500 90 90 100 6
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature SERVO_TILT
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
[... 405 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 18
set pitch_rate = 18
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

krzysztofmatula, 2017-04-19:
It's not normal behaviour, however I too observe it since some time (since inav 1.2?)
Maybe not that the quadcopter is trying to start, but motors are definitely not idling as I'd expect.

digitalentity, 2017-07-09:
Behavior is normal if you are arming in ANGLE mode. Quad is never 100% level and I-term compensates for that.

closed 2017-07-09 by digitalentity

Xe54, 2017-07-10:
Hi, but this is never happened with Cleanflight on it, why? I think it is a bit strange.