iNavFlight/inav #1567: Target Request: SP Racing F3 Neo

gfacek, 2017-04-19:
Is it possible to migrate the SPRF3 Neo from Cleanflight or Betaflight to iNav?

oleost, 2017-04-19:
@gfacek If you bought it from Seriously Pro, maybe ask @hydra ?
Would guess it would be in his interest to have his boards working on INAV.

hydra, 2017-04-19:
Should be easy enough for someone to pull the commits or code from Cleanflight master.  If whoever does this has questions just ask.

digitalentity, 2017-04-19:
@hydra pulling over commits should indeed be fairly trivial. I might do it if I have some time to spare on this one.

gfacek, 2017-04-21:
This would be nice :)

gfacek, 2017-05-05:
Any news here?

martinbudden, 2017-05-05:
Yes, we should do this. It's just a question of having some spare time...

martinbudden, 2017-05-17:
@hydra , I've made a first cut at PR https://github.com/iNavFlight/inav/pull/1695 , can you review please? (`target.c` file needs careful checking.)

gfacek, 2017-05-19:
Nice! Thank you so long. Hopefully I can finally bring my copter in the air soon. :)

DzikuVx, 2017-08-28:
SPRF3NEO has been added some time ago, so I'm closing

closed 2017-08-28 by DzikuVx