iNavFlight/inav #1570: ivav and nrf2401

jamedavid2017, 2017-04-20:
some one  let me know this project working ?
Or should delete version INAV with NRF24L01  to people not confuse

digitalentity, 2017-04-21:
nRF24L01 was never in official builds - it is intended for developers who know what they are doing. nRF24L01 receiver is verified to be functional by myself using a CC3D board and Cheerson-CX10A transmitter.

digitalentity, 2017-04-21:
If you want to get help with making nRF24 receiver working, please consider adding more details about your setup.

jamedavid2017, 2017-04-21:
Yes . I tried connect nrf2401 to some fc naze32 , cc3d......use devo 7e  firmware deviation to chốe protocol  syma, cx10..... all protocols don 't work. Even it doesn't bind . 
You can show me a video that you worked with it ? 
Thanks

martinbudden, 2017-04-21:
The NRF24L01 stuff does work, but the work on it has stalled a bit - I've been busy with other things. It also needs to be documented better.

However I am keen for others to try using it. Give us some more details about your setup and I'll try and help.

jamedavid2017, 2017-04-22:
hi martinbudden 
how to bind INAV + nrf2401  with protocol v202 ? it is a uto bind ??
i use FC NAZE32  with connect :
NRF24_CE_PIN PA1
 NRF24_CSN_PIN PA6
 NRF24_SCK_PIN PA7
 NRF24_MOSI_PIN PB1
 NRF24_MISO_PIN PB0

thanks

jamedavid2017, 2017-04-25:
????

jamedavid2017, 2017-05-09:
??????

DzikuVx, 2018-04-04:
issue appears to be dead, closing. Thanks

closed 2018-04-04 by DzikuVx