iNavFlight/inav #1572: MS5611 and HMC5883 not detected in SPRacing F3 Deluxe

cimenx, 2017-04-20:
In Betaflight It can Auto detected and read fine. But in Inav its not auto detected.
Also, when I set manually `set baro_hardware = MS5611` the CPU load rise to about 660%
[image]

digitalentity, 2017-04-21:
Please consider providing details of your setup, complete dump and bootlog.

rb1205, 2017-04-24:
Just tested a SPR F3 deluxe on both 1.6.1 and 1.7.0 RC1. Both version correctly detected and use the available hardware.

digitalentity, 2017-04-25:
I guess it's a hardware quality issue. The fact that sensor is being detected by Betaflight doesn't mean that it's functioning properly. INAV is much less tolerant to faulty sensors.

dragon1020, 2017-04-25:
Have same issue. Old F3 Deluxe was works fine, but with new got problem, which described by cimenx. And yes, in BF all ok - at least on graphic on "Sensors" tab magnetometer and barometer works like a charm.
Interesting thing: this sensors works just about fine in iNav 1.5.0. In 1.5.1 they don't. In 1.5.0 only one issue - magnetometer randomly resets.
[image]

digitalentity, 2017-04-25:
Neigher Betaflight nor INAV before 1.6 have hardware failure detection. INAV 1.6 and above will detect and disable failing or unreliable sensors for safety. I would assume everybody will want to know about sensor failure before taking off.

For the record, both baro and compass work reliably on my **genuine** SPRacingF3 board.

dragon1020, 2017-04-25:
And why iNav 1.5.1. didn't work as well? Only 1.5.0 (and probably below).
Not all of us can buy genuine F3. So we can't fly?

digitalentity, 2017-04-25:
There is no difference in 1.5.1 and 1.5.0 regarding i2c bus operations on F3 targets (including SPRacingF3). The board should work equally good (or bad) with both firmware versions. Most likely you are experiencing random sensor failure that manifest itself on all firmwares.

Quality problems are very frequent with Chinese boards - you should contact the seller of your board for solving quality issues.

Regarding your question wether you can fly or not: in the end it's either blocking faulty boards from flying or having tons of "fly-away" reports. To me choice is pretty much obvious,.

dragon1020, 2017-04-25:
Random sensors failure should be _random_, right? :) But I (and one more dude, check your VK group, please) get working (almost) board at 1.5.0, and it hangs with over 600% CPU at 1.5.1. I tested more than once. Something has done between 1.5.0 and 1.5.1.

olivierlando, 2017-04-26:
I'm having exactly the same issue with an SPF3 board : MS5611 & HMC5883 are detected and seem to be working fine with iNav 1.5.0. In the following versions, they are not detected and if manually select them, CPU load go up to 600%.

dragon1020, 2017-04-26:
olivierlando, welcome to the club. ;) With you I know 5 peoples with this problem. Where you bought your board?

digitalentity, 2017-04-26:
Nothing affecting I2C and sensor detection on SPRacingF3 board was done between 1.5.0 and 1.5.1. The glitch is triggered by something else. I probably should buy one board from the same source as you guys to investigate.

dragon1020, 2017-04-26:
Konstantin, I (and other guys) is ready to help as we can. Just say what to do. :) May be this issue didn't link to I2C?

digitalentity, 2017-04-26:
It's virtually impossible to debug remotely. I would use logic analyzer to check the activity on I2C bus to see what sensor locks up the bus.

dragon1020, 2017-04-26:
May be I will send to you my board, when I get refund for it? May be this will be faster, than order it from China?

digitalentity, 2017-04-26:
@dragon1020 where are you located?

dragon1020, 2017-04-26:
Moscow, Russia. :)

olivierlando, 2017-04-26:
I bought mine on Amazon : https://www.amazon.fr/gp/product/B01D8EK1YY . I've made some more search and it seems that there are often I2C issues with this version of the board. In Cleanflight the sensors are detected but I can see an lot of I2C errors... I'm going to ask for a refund. Maybe these blocking errors in iNav have saved my plane ;)

digitalentity, 2017-04-26:
That's the point of hardware failure detection - you want to know about the failure **before** you fly :laughing:

digitalentity, 2017-04-26:
@dragon1020 yes, it would definitely be faster than ordering from China. Ping me on VK - once I have failing hardware we'll get this squared out - maybe there is some hardware or software solution to this issue.

dragon1020, 2017-04-26:
Ok, but I bought mine on Ali, from a famous :) supplier ReadytoSky.

digitalentity, 2017-04-26:
ReadyToSky is know for random quality but very cheap hardware. Finding a decent quality hardware has become a huge problem lately.

darkling1979, 2017-04-27:
Same problem, SPRacingF3 Deluxe from ReadyToSky, 
versions <= 1.4 mag works fine, i2cErrors=0
version=1.5 mag does random jumps to zero(one or two per 10 sec),i2cErrors=0
versions >1.5 processor load =600%, i2cErrors over 9000.
I used internal and external mag, it makes no difference in described system behavior.

georges6788, 2017-04-28:
Similar issues on my board SPRF3 Deluxe which I bought it in UK.

digitalentity, 2017-04-29:
SPRacingF3 from a random Chinese seller running 1.7.0 for 5 hours straight. No issues.

https://cloud.githubusercontent.com/assets/11059099/25552814/c01a7e14-2ce6-11e7-99fd-37e855f0fbb1.png

For troubleshooting this I need either a failing board or a recording of SCL and SDA done with a logic probe at the moment of failure.

Tylerrattv, 2017-04-29:
im having the same issue. if i manually enable the compass it still works but as soon as the barometer is enabled the CPU goes up to 660%

liutairan, 2017-05-01:
Well, I got exactly the same problem with my new SP Racing F3 board. I tested with cleanflight/betaflight and iNav 1.5.0 all work well, but got problem with iNav 1.5.1, 1.6.0, 1.6.1.

liutairan, 2017-05-01:
@digitalentity I also bought my SP Racing F3 board from the amazon link appeared above (https://www.amazon.fr/gp/product/B01D8EK1YY) as @olivierlando .

 I will return it and find another one.

digitalentity, 2017-05-01:
Name and decription of the board on Amazon (especially the "Brano" word - should be Baro or Barometer) give a hint that it's a cloned board meaning there are probably cloned, faulty or incorrect components used.

What I suggest you try doing is add a pair of 1K pull-up resistors between SCL/SDA and 3.3V. You can take any other source of 3.3V on the board, just be sure to verify it's 3.3V or you might burn the hardware:

[image]

georges6788, 2017-05-01:
I've tried with those two resistors - still the same; I'm pretty sure the behavior of my board is not as it should be; again, seems to be fine in Cleanflight/Betaflight but 
I've flashed  iNav firmware through iNav configurator (for this test) when tried to come back to Cleanflight firmware just stopped working - had to use ST flash loader to get it back to life.
I've ordered a new SPRF3 board from another source to test it.

georges6788, 2017-05-05:
The new SPRF3 (from trusted source this time) just arrived; flashed iNav firmware 1.7.0 which works fine; no errors and CPU load 5 - 12%.
I hope this helps.

digitalentity, 2017-05-05:
I believe this issue can be closed - obviously hardware-related.

dragon1020, 2017-05-05:
Not too early? You don't look at my FC yet. And by the way, look private messages in VK - it's about sending to you my FC.

moonlitfpv, 2017-05-12:
georges6788; where did you get your board from in the end? I have a Chinese board that's acting out the same as everyone elses. Genuine SP board then?

liutairan, 2017-05-12:
@georges6788 I am also curious where did you get the new board; I am looking for a working SPRF3. Thanks~

georges6788, 2017-05-12:
@moonliftfpv, liutairan 
this FC is Chinese too however I bought a few things from this seller and I'm very happy with them.

http://www.ebay.co.uk/itm/262620641060?_trksid=p2060353.m2749.l2649&var=561516167389&ssPageName=STRK%3AMEBIDX%3AIT

Allerhopp, 2017-05-13:
2 new SPRacingF3 deluxe, i can confirm darkling1979:

> Same problem, SPRacingF3 Deluxe from ReadyToSky,
versions <= 1.4 mag works fine, i2cErrors=0
version=1.5 mag does random jumps to zero(one or two per 10 sec),i2cErrors=0
versions >1.5 processor load =600%, i2cErrors over 9000.
I used internal and external mag, it makes no difference in system behavior.

I have measured the internal pullups and there are 1k @ SCL and SDA. Is maybe 1k too low? Can someone with a working SPRF3 measure the pullups (between SCL <-> 3.3V and SDA <-> 3.3V)?

Allerhopp, 2017-05-20:
After removing the magnetometer chip (i have heated the chip on top with a soldering iron, until it was removable) , INav detected the baro and gives the expected values. So the magnetometer chip seems to be the culprit. No loss for wings and little loss for copters, they usually need a separate mounted magnetometer.

GTorreil, 2017-05-31:
@Allerhopp What FC did you modify this way ? SPF3 from Aliexpress / Readytosky ? If this works then I am extremely interrested !

GTorreil, 2017-05-31:
That's exactly the one I have !! Thanks a lot :D 
I'm only concerned about something... Will the mag in an M8N module from Aliexpress have the same problem? We'll see...

dawidmolenda, 2017-06-02:
Same problem. In cleanflight everything works reliably, some i2cErrors on startup.
iNav 1.4 works, 6 i2cErrors on startup.
iNav 1.5 random jumps to 0 and i2cErrors=0.
iNav >= 1.5.1 hangs.

darkling1979, 2017-06-03:
unsoldered an resoldered pullup resistors for i2c bus, and it5 started to work normally.(with external mag
[image]
)

myvilli, 2017-06-11:
Hi,
I am so happy to bump into this thread, I desperately tried all the possible settings and can only see my wing crashing down.
Nor horizon nor angle nor RTH mode was working correctly.
I am using SPF3 mini and Unavailable 1.7 while 1.4 is working flawlessly.
I think the i2c code has been modded with the introduction of RTH sanity check of Inav 1.5. Wouldn't be possible to set it as a parameter?

leper666, 2017-06-16:
Hi folks,

unfortunately I have  a faulty board too. Very bad beginning of my holidy:( However I have not that much time to wait for a new one. So I tried improvin solder joints on the i2c pull up resistors with no luck. MY board (v1) look a bit different from darklings but i manages to find the appropriate resistors using my multimeter. 
In my despair i was brave enough to remove the the HMC5883 and now everything is working like it is supposed to. Baro is auto recognized and I2c is working without errors.
So if you plan on building a gps controlled quad like me, the lesson is to save money and buy the 6DOF version in the first place...

Hope this helps someone!!! 

Best regrads from Germany.

Denis

leper666, 2017-06-16:
I have a short update:

I connected my neo 6m gps to the module. The onboard compass doesnot seem to work to at the moment.  
I'll have to sort out where the problem is tomorrow

VictorBucha, 2017-07-09:
Hi, 
I got the same problem with SP Racing F3 Deluxe, INAV 1.7.1. (also tested 1.7.0- same bad results) I cannot enter CLI in INAV as CPU load is above 650% and board is not reacting to any commands, but perfectly works with latest cleanflight or betaflight firmwares. I made dump in Betaflight for your reference below:
$M>ptARM;AIR MODE;ANTI GRAVITY;ANGLE;HORIZON;HEADFREE;BARO;MAG;HEADADJ;FAILSAFE;BEEPER;BLACKBOX;FPV ANGLE MIX;OSD SW;z$M>ù
`$M>nÌ¥
CLI

# dump
# Betaflight / SPRACINGF3 3.1.7 Apr  3 2017 / 22:27:23 (e1c4b5c)
name -
resource BEEPER 1 C15
resource MOTOR 1 A06
resource MOTOR 2 A07
resource MOTOR 3 A11
resource MOTOR 4 A12
resource MOTOR 5 B08
resource MOTOR 6 B09
resource MOTOR 7 A02
resource MOTOR 8 A03
resource MOTOR 9 NONE
resource MOTOR 10 NONE
resource MOTOR 11 NONE
resource MOTOR 12 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A00
resource PWM 1 A00
resource PWM 2 A01
resource PWM 3 B11
resource PWM 4 B10
resource PWM 5 B04
resource PWM 6 B05
resource PWM 7 B00
resource PWM 8 B01
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A14
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 B05
resource SERIAL_TX 12 B01
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A15
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 B04
resource SERIAL_RX 12 B00
mixer QUADX

mmix reset

servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
[... 433 lines cut ...]
Allowed range: -32768 - 32767

magzero_y = 0
Allowed range: -32768 - 32767

magzero_z = 0
Allowed range: -32768 - 32767

# get baro
baro_tab_size = 21
Allowed range: 0 - 48

baro_noise_lpf =  0.600
Allowed range: 0 - 1

baro_cf_vel =  0.985
Allowed range: 0 - 1

baro_cf_alt =  0.965
Allowed range: 0 - 1

baro_hardware = AUTO
Allowed values: AUTO, NONE, BMP085, MS5611, BMP280

#

VictorBucha, 2017-07-09:
I flashed Inav 1.4.0 and all sensors detected. 
dump is below:
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPRACINGF3 1.4.0 Nov 28 2016 / 10:59:44 (6bf7a91)
# pflags
# Persistent config flags: 0x00000000
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -UNUSED_1
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -UNUSED_2
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature RX_PPM
feature VBAT
feature FAILSAFE
feature BLACKBOX

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
[... 383 lines cut ...]
set acc_soft_lpf_hz = 15
set dterm_lpf_hz = 40
set yaw_lpf_hz = 30
set yaw_p_limit = 300
set iterm_ignore_threshold = 200
set yaw_iterm_ignore_threshold = 50
set rate_accel_limit_roll_pitch = 0
set rate_accel_limit_yaw = 10000

# dump rates

# rateprofile
rateprofile 0

set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

leper666, 2017-07-09:
In my opinion the problem with the high cpu load is that the i2c bus is notworking as inav ist expecting it to work. I solved the problem by removing the onboard magnetometer. I than had the same problem again after connecting an external magnetometer on my gps module via i2c and activating it. It turned out that the gps module had no supply voltage bevaus the SP3 does not output 5v when power over usb.
I connected the battery and then i had all sensors working. Somehow the flight performance was very poor. But the problem for that could have been a faulty ESC. I have to wait for the new ones for further testing.

Best regards,

Denis

krish-90, 2017-07-09:
I was having the same problems mentioned above until I desoldered the HMC5883 chip. As soon as I did that I was able to connect with the iNav configurator and the barometer is connected and working now.

VictorBucha, 2017-07-09:
So did I right now - desoldered  the HMC5883 after useless effort to power up the board from external source. now everything work fine. Mag will do better when distanced as far as possible from FC. 
This is definitely a hardware issue

dawidm82, 2017-07-12:
If this is a hardware issue then why it works fine in cleanflight/betaflight and iNav 1.4.0?

digitalentity, 2017-07-13:
Because neither CF not early INAV cares about sensor quality and reliability. If sensor is at least sporadically detectable - CF and early INAV consider it usable which in turn have caused a flood of "fly-away" bugreports some time ago.

MatteoPi, 2017-08-22:
I have same issue to sp3, baro don't work with 1.7x version, but it work with 1.6.1 version inav
For me there is a bug to new version 1.7.x

digitalentity, 2017-08-28:
Closing. See #1969 #1934 #724

closed 2017-08-28 by digitalentity