iNavFlight/inav #1573: Switch RTH to airplane mode in case of compass failure

brat002, 2017-04-20:
What do you think about next feature: in case of compass failure due RTH for multirotors instead of landing in unknown conditions switch flight mode to "airplane mode?. This mean - multirotor fly only in head and heading calculated only from GPS. After reach home location multirotor should initiate landing as usual behaviour for RTH.

rb1205, 2017-04-21:
Is a sudden compass failure a likely scenario? I had some problem with it not being detected when the FC reboots, but aside for that it never failed mid flight on me.

brat002, 2017-04-21:
I think this method of RTH is much safe for all autonomous flights. No matters what is the reason of this flight.

rb1205, 2017-04-24:
It changes a simple and prevedible straight line flight into a potentially long turn, increasing the space and time it takes to return. Also what are you going to do when the multicopter reaches the home position and starts to descend? The FC needs the compass to apply GPS hold. It'd be an emergency landing.

digitalentity, 2017-04-24:
Quad can't do a coordinated turn without knowing what direction it is facing. During a turn it will drift sideways which may lead to a situation where it is already facing new direction of flight but still flies in old direction (sideways). Therefore I think that switching to "hybrid" RTH mode is in fact less safe than landing where it is. Also, it's always safer to make sure compass if functioning properly.

closed 2017-04-25 by brat002