iNavFlight/inav #1577: INAV 1.7.0-RC1: OSD failures

raul-ortega, 2017-04-21:
- Startup message doesn't appear (INAV version, and how to enter to osd are missing).

- ARMED message doesn't appear when arming (from switch).

- Statistics doesn't appear when disarming (from switch).

- Can't enter in osd menu with THR MID + YAW LEFT + PITH UP

digitalentity, 2017-04-22:
@raul-ortega what board and what target do you use?

raul-ortega, 2017-04-22:
Betaflight F3 AIO v1.1 target OMNIBUS.

With 1.6.1 everything about OSD works well.

digitalentity, 2017-04-23:
Interesting. I'm current traveling and can't verify and debug. @DzikuVx do you still have Omnibus F3 board? Can you confirm the issue?

raul-ortega, 2017-04-23:
Yes, this is my actual board (the Banggood clon), I fly my s800 sky shadow
with it. I confirm the issue.

DzikuVx, 2017-04-23:
I should have one somewhere. Mini, not regular version, but the same target

Soph, 2017-04-25:
I can confirm I have the same issue. Omnibus F4 PRO

Soph, 2017-04-25:
I looked a bit at the code and the commits since 1.6.1. If I checkout 1.7.0-RC1 and revert this commit

https://github.com/iNavFlight/inav/pull/1506/commits/0f2138ca4781e0a3d3bd24a821f0c315cbec2f8a

The stick commands make the menu appear again. But then it's not responding to any input.

DzikuVx, 2017-04-26:
@raul-ortega @Soph could you please share CLI dump in which OSD is not working like expected?

Soph, 2017-04-26:
@DzikuVx here is the dump for my setup. Flashed to "stock" 1.7.0 RC1 before doing the dump. https://gist.github.com/Soph/520ee8617bf8b72fded57c4126a79f0c

DzikuVx, 2017-04-26:
@Soph OK, I will try to flash my omnibusf4Pro today

Soph, 2017-04-26:
@DzikuVx for the Omnibusf4pro: To debug on the device, does this still apply: https://github.com/iNavFlight/inav/blob/master/docs/development/Hardware%20Debugging.md

Or is there an easier way to debug straight trough the USB connection?

DzikuVx, 2017-04-26:
@Soph I think this is the only way to debug.

@marbalon can we have your assistance on the topic of CMS not working like expected? You seems to know what is happening down there

raul-ortega, 2017-04-26:
Here you are my dump and diff configuration files.

[inav_1.6.1_S800_SkyShadow-diff-LAUNCH-ARM-AUTOTUNE-SWITCHES.txt](https://github.com/iNavFlight/inav/files/959502/inav_1.6.1_S800_SkyShadow-diff-LAUNCH-ARM-AUTOTUNE-SWITCHES.txt)
[inav_1.6.1_S800_SkyShadow-dump-LAUNCH-ARM-AUTOTUNE-SWITCHES.txt](https://github.com/iNavFlight/inav/files/959501/inav_1.6.1_S800_SkyShadow-dump-LAUNCH-ARM-AUTOTUNE-SWITCHES.txt)

Soph, 2017-04-26:
Since we both use the FLYING_WING mixer (and a SkyShadow ;) ) I just tried if it's related to that because it's not using YAW at all. But switching to a quad preset didn't change anything.

raul-ortega, 2017-04-26:
But with version 1.6.1 works!!!

digitalentity, 2017-04-29:
Fixied.

closed 2017-04-29 by digitalentity

raul-ortega, 2017-04-29:
Tested, startu up message is shown, and you can enter to the MENU, but  **it is not possible to enter in the menu options,  neither SAVE or EXIT**.

digitalentity, 2017-04-29:
@raul-ortega I've verified that menu works fine before closing this. Tested on Omnibus F4 Pro.

raul-ortega, 2017-04-29:
Ok, tested again and it works. Some minutes before I was testing it on a multicopter with NAZE and external OSD, and I applied YAW LEFT instead of PIT LEFT.

It was a confusion, I apologyze (too many owers testing, with different systems).

digitalentity, 2017-04-29:
@raul-ortega no worries, it's good to double check as I might have missed something :smile: