iNavFlight/inav #1579: ANYFCF7 : CPU load

urlu75, 2017-04-22:
with default firmware 1.7.
Racer 5" racer configuration applied and then just change gyro, acc attitude and loop time to maximum. CPU load is >400%. Can be more than 600% if some features are enabled.

```
# status
System Uptime: 166 seconds
Voltage: 0 * 0.1V (3S battery - OK)
CPU Clock=216MHz, GYRO=MPU6000, ACC=MPU6000
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, SONAR=NONE, GPS=NONE
SD card: None inserted
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 1, config size: 3271, max available config: 262144
System load: 425, cycle time: 141, PID rate: 7092, RX rate: 24, System rate: 9
```

I2G mag issue still there with 1.7.

stronnag, 2017-04-22:
Not sure what problem you're reporting here. 

If you enable features beyond that which the CPU can support, it does not work. So what do you expect?

urlu75, 2017-04-22:
So you are saying that it is normal that on on an F7 board, CPU is load is > 400% just with 
- gyro LPF cutoff 256Hz
- sync mode : all
- gyro task : 8kHz
- acc task : 1kHz
- attitude : 1kHz
- FC loop time : 8kHz

Same configuration on an F4 board, CPU load is only 35%

sambas, 2017-04-22:
All the performance updates from BetaFlight has not yet landed here

digitalentity, 2017-05-18:
@sambas what's the status of this? Fixed or should we postpone to 1.8?

sambas, 2017-05-18:
postpone, not sure why pid controller isn't happy when using 8/8

digitalentity, 2017-05-18:
Aye!

digitalentity, 2017-10-08:
@sambas any news regarding this?

DzikuVx, 2017-10-11:
I just tested AnyFCF7 with 8/8kHz, acc at 1kHz and attitude and 500Hz. Load at 37%, I suppose we can assume problem solved. Thank everyone

closed 2017-10-11 by DzikuVx