iNavFlight/inav #1582: FW - RTH altitude not capped

PaulHam211, 2017-04-22:
I tested RTH today and noticed that while circling above the home point my plane was gaining altitude and not staying at the nav_rth_altitude which I set to 6000 (60m), it finally got up to 113m before I cancelled RTH.
I worry that if I lost radio control and the plane returned home, it would get stuck circling the home point and end up miles high. 

If you watch when RTL is engaged, the plane climbs to 60m and then turns. On the turn, the plane gains 13 meters before dropping back down to 60m when flying straight.

It seems that when turning, my plane gains alt, in RTH the altitude is not capped at the nav_rth_altitude setting?

Heres a video of the RTH portion:
https://youtu.be/gFci_HR7W0U

digitalentity, 2017-04-22:
If you have blackbox recording of the flight - please share.

PaulHam211, 2017-04-22:
@digitalentity  Thanks for your reply.
I do have some logs, I am new to logs so please bear with me.

I do not know which log relates to this clip, but I think I did try a RTH on each flight today.

Also, my final flight ended in a crash. I was coming into land and lost video, when the video came back I was on the ground and the flight controller had rebooted (perhaps a power issue) I don't know if this shows in the last log or if it got cut out.

[LOG00022.TXT](https://github.com/iNavFlight/inav/files/949361/LOG00022.TXT)
[LOG00023.TXT](https://github.com/iNavFlight/inav/files/949364/LOG00023.TXT)
[LOG00024.TXT](https://github.com/iNavFlight/inav/files/949362/LOG00024.TXT)
[LOG00025.TXT](https://github.com/iNavFlight/inav/files/949363/LOG00025.TXT)
[LOG00026.TXT](https://github.com/iNavFlight/inav/files/949366/LOG00026.TXT)
[LOG00027.TXT](https://github.com/iNavFlight/inav/files/949365/LOG00027.TXT)

MaxFpv, 2017-05-02:
I have the same Problem that over the homepoint altitude is raising

digitalentity, 2017-05-26:
I think this issue will require re-testing when 1.7.1 is out in a few days.

schnupperm, 2017-06-03:
I have the same problem on my 1.6.1 z-84 wing wing. The wing climbs during pos hold while turning. (it seems not related to RTH)
[LOG00004.TXT](https://github.com/iNavFlight/inav/files/1049497/LOG00004.TXT)

digitalentity, 2017-08-28:
Is the issue still relevant?

krzysztofmatula, 2017-08-28:
It's hard to say. It happened to me only once and I was unable to reproduce it again neither on the same nor on any newer firmware.
I propose to close the ticket for now...

digitalentity, 2017-09-04:
Closed due to no further reports of the issue

closed 2017-09-04 by digitalentity