iNavFlight/inav #1586: Cycle Time going crazy when using MSP at 9600bauds (EZ GUI & Configurator) leading instability

giacomo892, 2017-04-23:
The machine is running INAV 1.6.1 on NAZE target with MSP running at 9600 attached to a software serial but I have tried the same with 1.7RC1 on another machine without blackbox logging but withan hardware serial and it show the same behaviour. 

Basically I've noticed the machine was acting weird at the field in the way it was loosing attitude for fractions of seconds once in a while.

Then on the 1.7 machine i've also noticed that the MSP stopped working after landing and I was unable to arm the machine again (LEDs where flashing when I tried to arm) so it seems like the machine was not able to pass some pre-arm checks (CPU load?)

So I went home and tried to track the issue down. Removed all the props and attached the machine to the configurator. 
The loop time stayed very close to 2000 when the configurator was running on the setup tab but as soon I switched to Motor tab the magic arrived.

Loop Time reported in the low part of the configurator started to jump (even to 25000) and I can hear the motors acting erratically (spikes) some times.

Then done anothers trials with blackbox logging enabled and one in particolar scaried me a lot.

Check the (5/5) part of the attached log.

I turned the remote control off to simulate a failsafe RTH leaving motor tab opened on the PC and the CPU load /cycle time reported went completely crazy.

I did and empirical measurement from the log taking the loop iteration index and the time:

Iteration:  13135               18156     
Time:    144354484            160691272

This gives and average of 3253uS loop time and it should be 2000uS

Excuse me if I wrote too much and maybe forgot to mention something but the log should speak by it's own expecially part 5. Also notice the rcCommand moving by it's own the remote is off, is this normal?

[LOG00198.TXT](https://github.com/iNavFlight/inav/files/949569/LOG00198.TXT)

stronnag, 2017-04-23:
1. Are you using MSP sanely, i.e. send a request, wait for a response? In using MSP / 9600 on NAZE since pre-1.0, I have never seen this behaviour. 

2. Do you see the same behaviour in the configurator if you set 'Wireless' mode (which is specifically designed to avoid overloading the CPU at low bandwidths).

giacomo892, 2017-04-23:
1)Bart (EZ-GUI) told me that now the app tries to keep a constant number of frames in queue. Maybe it still too much.

2)Using wireless mode on the configurator does the trick but there should be a protection on FC's MSP side to avoid to overload the FC and "crash" it, since in my previous trial it wasn't able to recover even if I stop sending MSP requests.

As for avoiding any issue at the moment I strickly stick to LTM telemetry. With it and blackbox logging enabled i've checked with the same method as before the loop time is 2000uS on average :)

stronnag, 2017-04-23:
I doubt FC can protect against this kind of abusive behaviour. It requires a cooperative  application (mwp, configurator with 'wireless')

For example, I modified one of the mwp cli test tools to (a) use synchronous I/O (in a loop, send MSP_STATUS, get response, repeat), and (b) use asynchronous I/O (send MSP_STATUS every 20 milliseconds).

Results (for one minutes, min,max are cycle times):

**Sync** : 60 secs, iterations 1858, min=2000, max=2364 sync
**ASync** : 60 secs, iterations 2860, min=2000, max=21322

So it's quite obvious;  use well behaved applications if you want safe FC behaviour. Spamming the FC with asynchronous request for a synchronous protocol is beyond documented behaviour.

digitalentity, 2017-04-23:
We can discard everything coming via serial link until MSP response is completely sent out. It's a hack, but it might protect FC from ill-behaving applications.

giacomo892, 2017-04-23:
@stronnag  thanks for your testing but as DE says it is better that FC will protects by its own.

It can be a good thing @digitalentity also maybe with CLI command that enables this functionality  but also we might want to limit MSP output rate like maximum x resposes/second.

Input rate shouldn't be a problem since if the app start spamming the input buffer will overflow and it will kind of autolimit the rate as far I can tell.

eziosoft, 2017-04-23:
At the moment ez-gui sends msp and waits for return at startup and for important stuff. For less important data it sends 4 or 5 msps and waits for returning of the last one. I've tried it on slow links and it worked well.

stronnag, 2017-04-23:
It will be interesting to see how much the proposed fix for this 'problem' breaks

giacomo892, 2017-04-23:
Maybe we just need to limit the time the FC can spend inside mspSerialProcess() and not processing the whole queue everytime the FC gets there (100 times per seconds).

eziosoft, 2017-04-23:
As I remember correctly multiwii processed one msp per cycle. I don't know if that limit is also in inav.

giacomo892, 2017-04-23:
From what I can understand INAV processes the whole serial queue 100 times per seconds and maybe it can be too much.

digitalentity, 2017-04-24:
@giacomo892 INAV only processes one MSP command per cycle, the problem is that MSP response is usually quite bigger than request, so we have an output queue full and firmware blocks waiting for it to be free enough to push another MSP response. A classis "Denial of service" attach :laughing: 

If possible, can you please test https://github.com/iNavFlight/inav/pull/1593 ?

eziosoft, 2017-04-24:
So what would be ideal scenario to send msp requests? 
From my experience sending one request and waiting for replay and send another one doesn't work good with 3dr radio. It is very very slow. My guess that is because 3dr must switch between transmit and receive.

eziosoft, 2017-04-24:
Is this still a problem when you enable wireless mode in configurator ?

giacomo892, 2017-04-24:
@digitalentity Tomorrow i'll build the firmware and test. 
@eziosoft No, enabling wireless mode all is ok.

stronnag, 2017-04-24:
Using https://github.com/iNavFlight/inav/tree/de_msp_denial_of_service_fix /  #1593

msp torture test (async = 1 message every 20ms), typical results:

**Sync** : 60 secs,  sent=2014, recv 2013, min=2000, max=2342
**Async**: 60 secs,  sent=5867, recv 2834, min=2000, max=2360

Tested on flip32, baro, mag and gps enabled. softserial 9600 baud.

ShikOfTheRa, 2017-04-24:
Also interested for MWOSD.
by default main loop sends req every 20ms if not received response it resends same req until response received then sends next msp req.

giacomo892, 2017-04-25:
Already commented on the pull requests. Everything it is sane from my point of view.

DzikuVx, 2017-05-17:
OK,. since #1593 has been merged, closing this one

closed 2017-05-17 by DzikuVx