iNavFlight/inav #1590: Aerial 3D mapping

LucasFPV, 2017-04-23:
Hello, 
Would be great to add the feature in INAV with mission planner to take at selected moments a picture from a camera (servo output to send signal to the camera) to then create a 3D Aerial Map. 
Something similar as the APM feature would be great!
Regards,
Lucas FPV  
Example: https://www.youtube.com/watch?v=2ZsO3LWtd9g

digitalentity, 2017-07-09:
Probably won't happen any time soon.

oleost, 2017-07-09:
Could it be implemented simple?

Take picture every X second on Hardcoded to channel 20 when featue is enabled. Then to use you simple forward CH 20 to an aux channel. ( Taking pictures just mean taking CH20 to 2000us )

Could addiotionl be configured to take picture every waypoint instead of time based.

LucasFPV, 2017-07-09:
Yeah, we could set a switch to toggle during flight and with an already planned Mission planner route, it would activate via an RC servo output pin for example the gorpo shutter each Number of seconds. Found this stuff which is compatible with the gopro wifi to convert an rc servo input to activate the shutter. https://www.youtube.com/watch?v=t-QeQ9M0KKg

oleost, 2017-07-09:
Btw if you own an advanced radio like taranis you can simple do this by
radio, program it go high every x second while switch is active.

9. jul. 2017 4.07 p.m.

or mute
the thread

LucasFPV, 2017-07-09:
Got an advanced radio but not good for FPV.. Futaba T18SZ

nm17, 2020-06-02:
Any updates on this? Still need this feature

myrcwing, 2021-07-19:
Following up to see if anything was updated for camera trigger in INav 3.0. Any plans for this to be added?

nm17, 2021-08-16:
It would be great to have at least more info on external camera trigger technology. I'll try to connect my old digital camera to my drone and try to figure out stuff, **although** if anyone is reading this and is trying to try to do something like that themselves, don't rely on me delivering, I might forget about this issue again, maybe not.

nm17, 2021-08-16:
@myrcwing can you provide a way to contact you? My email is `nm17@riseup.net` and it would be great to know yours, so we could figure things out together.

nm17, 2021-08-18:
@LucasFPV you too

> @myrcwing can you provide a way to contact you? My email is `nm17@riseup.net` and it would be great to know yours, so we could figure things out together.

closed 2023-10-27 by DzikuVx