iNavFlight/inav #1592: Don't use 280 mm tricopter preset!

jstremmler, 2017-04-24:
I used the 280 mm tricopter preset in iNav 1.6.1 just as a trial for my Trifecta tricopter from HK!

It immediately destroyed my tail servo! So don't use it!

 I really don't have any understanding how this setup  has been implemented in iNav 1.6 without any warning for a possible servo damage.

oleost, 2017-04-24:
Want to elaborate "destroyed" my servo?

The preset already mentioned it is for digital servos. Did you try and use it for analoge?

digitalentity, 2017-04-24:
@jstremmler please share details of your setup. If you used an analog servos and didn't pay attention to preset description - it's obviously not a firmware or preset fault. Firmware and Configurator have no idea what equipment you were using.

jstremmler, 2017-05-02:
Ok it was my mistake! 

I overlooked in the top field in presets the "digital servo" and didn't have in mind that the required tail servo for the HK Trifecta, the TGY-9018MG, is an analog one.
So here is what I did in the meantime because the TGY 9018 is/was the one and only servo which does fit  in the gearbox of the Trifecta:

I designed with TurboCAD a gearbox for the Tifecta which fits on a fast 27 teeth  digital servo for the tricopter, printed that in ABS, published the gearbox on Thingiverse and repaired my trcopter with a digital servo.
And now everything is ok.... and this issue can be closed.....

stronnag, 2017-05-02:
Cool, you will not regret the better servo.

DzikuVx, 2017-06-09:
I'm closing this one and creating a ticket in Configurator project to lower servo rate to default 50Hz for safety reasons
iNavFlight/inav-configurator#206

closed 2017-06-09 by DzikuVx