iNavFlight/inav #1595: Strange behaviour on failsafe recovery

brat002, 2017-04-25:
There are regression in 1.7. 1.6.1 was not affected this.

Inav 1.7 RC
Sparky 2
dsmx satellite

On rc failsafe recovery a plane becomes uncontrollable and just falls to the ground. This happens 3-5 seconds and then plane restores control and stabilisation.

`
# dump

# version
# INAV/SPARKY2 1.7.0 Apr 19 2017 / 09:32:55 (65cbad0)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1400 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature RX_SERIAL
feature GPS
feature TELEMETRY
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
[... 407 lines cut ...]
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 19
set pitch_rate = 19
set yaw_rate = 20
set tpa_rate = 25
set tpa_breakpoint = 1300
`

digitalentity, 2017-04-25:
Do you have logs?

digitalentity, 2017-04-25:
Also, can you describe a bit more what happens after signal is lost and what happens after signal is restored.

brat002, 2017-04-25:
No any logs, sorry.
How to reproduce the problem:
1. Fly straight.
2. Swith off RC.
3. A Plane correctly switch to RTH.
4. After RC switched on a plane switch to previous flight mode, but flight controller dont give  control to pilot. Just dive and fall
5. After 5 seconds pilot can take control the plane and recover it from fall.

digitalentity, 2017-04-25:
In your case after you switch RC back on and don't move the sticks plane should continue RTH. Can you confirm that the plane instantly switched out of RTH when you switched on the remote?

Also, what mode were you flying before activating failsafe?

oleost, 2017-04-25:
I did several test on the build I have on my airplane, ( cc9e407 ) 

Failsafe worked as expected.

brat002, 2017-04-25:
> Can you confirm that the plane instantly switched out of RTH when you switched on the remote?

Plane switched out of RTH when RC was switched on and I move sticks. But instead of fly straight plane started to dive. Control was regained only after 5 seconds. I performed this test 2 times with identical results.

digitalentity, 2017-04-25:
I looked through the code and don't see where the 5s delay may originate - control should be released to pilot immediately.

@brat002 Sparky2 should be able to record blackbox logs to onboard memory. If possible can you record a log of the issue - would certainly help figuring out what the issue is.

oleost, 2017-04-25:
Stick threshold will lock throttle and yaw control from pilot until he moves roll/pitch

brat002, 2017-04-25:
Guys, one correction. Plane starts to dive in RTH mode when RC switched on again. RTH switched back to angle only after delay. Between **rc on** and switch to **angle** plane falls around 5 seconds.

oleost, 2017-04-25:
@brat002 Please provide blackbox log, I'm unable to reproduce your issue.

digitalentity, 2017-04-25:
We shouldn't pay too much attention to this issue though. I'm reworking the receiver/failsafe system. The new code won't make it to 1.7, but I would like to have it in 1.7.1 shortly after 1.7.

brat002, 2017-04-25:
I will try to record logs asap, but bad weather doest not allows to do it today. Maybe tomorrow will try again.

brat002, 2017-04-25:
I can't explain this but problem was solved itself. Today's evening we tested this plane and RTH worked as should. 
Please set low severity to this task or close it.

digitalentity, 2017-04-26:
I've added it to 1.8 milestone. We'll likely close it since 1.8 will have a huge overhaul of failsafe system

digitalentity, 2017-05-18:
I'm closing this one. @brat002 feel free to reopen if problem repeats itself.

closed 2017-05-18 by digitalentity