iNavFlight/inav #1597: ALTHOLD mode never sent in LTM when MAG mode is enabled

krzysztofmatula, 2017-04-25:
I think the MAG mode is usually a long-term setting, while ALTHOLD rather short-term.
I propose changing the order of if-s in ltm.c, so ALTHOLD is reported first, then MAG mode only if ALTHOLD is not enabled.

Is:
```
    else if (FLIGHT_MODE(HEADFREE_MODE) || FLIGHT_MODE(MAG_MODE))
        lt_flightmode = 11;
    else if (FLIGHT_MODE(NAV_ALTHOLD_MODE))
        lt_flightmode = 8;
```
should be:
```
    else if (FLIGHT_MODE(NAV_ALTHOLD_MODE))
        lt_flightmode = 8;
    else if (FLIGHT_MODE(HEADFREE_MODE) || FLIGHT_MODE(MAG_MODE))
        lt_flightmode = 11;
```

I'll prepare PR if you agree.

digitalentity, 2017-04-26:
Sure, please go ahead.

krzysztofmatula, 2017-05-03:
Fixed in 1.7.

closed 2017-05-03 by krzysztofmatula