iNavFlight/inav #1605: Question: shouldn't I2C errors also trigger Hardware Failure condition?

krzysztofmatula, 2017-04-27:
As some number of I2C errors appear during initialization phase we may reset the counters after that, or calculate the error increase within some time frame (e.g. each couple of seconds). 

After initialization I2C errors should not normally happen and pilots should be interested in getting visual and audible information that something is not working as it should.

digitalentity, 2017-04-28:
Good point. I think we should consider reporting it as CPU hardware health.

giacomo892, 2017-05-02:
I've another question related to I2C.
Our sensor are red by using bool i2cRead(I2CDevice device, uint8_t addr, uint8_t reg, uint8_t len, uint8_t* buf) function. 

In case of I2C error what will be the content of buf?

krzysztofmatula, 2017-05-02:
When the operation fails (returns 'false') the content of the buffer may contain garbage and shall not be interpreted. As a result a sensor may get into 'hardware failure' state, but it may stay there for a fraction of second only if the problem is not permanent. I'm proposing to use the information we have here as we do have error counter in I2C driver level, while not having such counters on sensors' level.

giacomo892, 2017-05-02:
Ok. 
But that fact that I really miss is if "garbaged" buffer gets returned (and it seems like) or not to the function that requires data ( read of gyro,acc, mag, baro..)

closed 2018-05-27 by stale[bot]