iNavFlight/inav #1613: Development version: "Navigation is safe" without GPS fix.

raul-ortega, 2017-04-29:
I apologyze if this is not the right place, but I'm not sure if this is a ina-configurator issue or a fimware issue.

The GPS has **no FIX** and **nav_extra_arming_safety** set to **ON** the pre-arming checks are all green, and I **can arm** (see picture).

[image]

My TARGET is OMNIBUS (ombnibus f3 aio v1.1 Banggood).

[inav_1.7.0-RC1_S800_SkyShadow-dump-LAUNCH-ARM-AIR-SWITCHES.txt](https://github.com/iNavFlight/inav/files/965995/inav_1.7.0-RC1_S800_SkyShadow-dump-LAUNCH-ARM-AIR-SWITCHES.txt)

giacomo892, 2017-04-29:
Are you sure you have enabled GPS in the configuration tab?

raul-ortega, 2017-04-29:
Yes, of course. Yes, of course. Have you taken a look to the dump file and the pre-arming chek picture?

raul-ortega, 2017-04-29:
I have tested it just now with 1.6.1 and it works as it should. Therefore, I guess it is a firmware problem (development version).

digitalentity, 2017-04-29:
You don't have any navigation GPS-aware modes enabled. There is no chance that GPS will be required so firmware is not enforcing the GPS fix.

raul-ortega, 2017-04-29:
I don't understand!
Thas this mean that when nav_extra_arming_safety is on and I don't have the home position set, I should be able to arm?
Should I activate the PS mode (I have it configured RTH and PH in aux channels)...

digitalentity, 2017-04-29:
You don't have PosHold or RTH modes configured in your dump. Please check that your modes tab is set up correctly.

digitalentity, 2017-04-29:
Extra arming safety enforces GPS fix only if there is a possibility that GPS would be required - if PosHold, Waypoint, RTH of Failsafe-RTH is configured.

raul-ortega, 2017-04-29:
Ups!!!! You are right, my dump seems to have errors (missing modes) when I change from version 1.7-RC1 to the the development. I really don't know what's happening, if I am doing something wrong...Does the development version override the configuration of the 1.7.0-RC1?

digitalentity, 2017-04-29:
Actually no, `aux` should be compatible between 1.6, 1.7 and development.

closed 2017-04-29 by raul-ortega