iNavFlight/inav #1617: Testing 1.7.0 RCs

stronnag, 2017-04-29:
A little belated, but a place for reports of successful (or otherwise) tests of 1.7.0 Release Candidates

stronnag, 2017-04-29:
**Commit**: b69f0a23
**Baord**: SPEVO
**Aircraft**: 600mm tricopter
**Results**:
Successful. Free flight, PH, WP, RTH. LTM not affected by recent changes.

stronnag, 2017-04-29:
**Commit**: 8c39f12a (+ de_msp_denial_of_service_fix + #1600)
**Board**: OMNIBUS F3 AIO
**Aircraft**: 420mm quad
**Results**:
Successful. Free flight, PH, WP, RTH. No observed difference (+ve or -ve) for #1600),

stronnag, 2017-05-01:
**Commit**: 10b9250e
**Board**: SPRF3 Acro + I2C baro / mag
**Aircraft**: 250mm quad / deadcat
**Results**:
Successful. Free flight, PH, WP, RTH. 

**Commit**: 10b9250e
**Board**: RMRC Dodo (SPRF3 firmware)
**Aircraft**: 280mm tricopter (no navigation sensors)
**Results**:
Successful. Free flight.

ahfynjj, 2017-05-02:
Commit: 
Board: OMNIBUS F4 Pro(V2)+ DJI GPS MAG
Aircraft: FX-61 flywing
Results:
Successful. ANGLE  HORIZON  NAV RTH

digitalentity, 2017-05-02:
I think we're ready for the final release. Other bugs would be fixed in 1.7.1 if necessary.

giacomo892, 2017-05-02:
Commit: INAV 1.7.1 RC1 + MSP FLOOD FIX
Board: NAZE
Aircraft: Custom Quad
Results: All working. ANGLE HORIZON AIRMODE ALTHOLD and LTM telemetry

isilcala, 2017-05-02:
Board: X-RACER v2.1 with SPRACEING F3 firmware + Beitian BN-200 GPS
Aircraft: FliteTest Tiny Trainer
Results: Successful. ANGLE AIRMODE STAB PASSTHRU RTH

digitalentity, 2017-05-03:
Thanks everybody!

closed 2017-05-03 by digitalentity

krzysztofmatula, 2017-05-03:
Just for reference - tested:

1. SPRACINGF3, 250mm quad, 6" - HORIZON, BARO, PH, RTH, FS, LTM - all ok.
2. SPRACINGF3EVO, 550mm quad, 10" - ANGLE, BARO, PH, RTH, LTM, CHANNEL_FORWARDING - all ok.