iNavFlight/inav #1618: Feature request: Addition of GTUNE

TwoToneEddy, 2017-04-29:
Is this possible to add without too much work? Would it be a case of copying it from cleanflight?

stronnag, 2017-04-29:
afaik, it is no longer if CF; it never really worked.

digitalentity, 2017-04-30:
If I recall correctly GTUNE never really worked. We do need a good autotune algorithm for quads though.

11digits, 2017-05-26:
What if we could have a different approach, a mode that would record data in blackbox for curent PID's and "auto" tune them after landing based on recorded data (even if it would offer some suggestions in the configurator like turn the P down or etc) ? Would that be possible maybe?

digitalentity, 2017-07-09:
GTUNE is not going to make it back. We will see to a different approach to autotune eventually.

closed 2017-07-09 by digitalentity