iNavFlight/inav #1619: Servo pins #6 and #7 not working for CUSTOMAIRPLANE mixer

Dejw4g, 2017-04-30:
I'm trying to use servo pins #6 for Flaps and #8 for retractable landing gear.

This is my smix:
smix 0 0 1 100 0 0 100
smix 1 1 0 100 0 0 100
smix 2 2 2 100 0 0 100
smix 3 6 10 100 0 0 100
smix 4 7 9 100 0 0 100

Elevator, Ailerons, Rudder and Throttle working just fine, I have a problem with last two definitions.

When I don't define them in SMIX (6 and 7) - these pins are just dead, even when I select respective AUX channel in Servos tab - still dead. I get a default signal out of them when I define smix for them as stated above (10 is AUX3, 9 is AUX2) but even though it looks like this in Motor tab: http://take.ms/n4EAn they won't move at all - they will just get into default position when power cycled (unlike when not defining them in smix). I tried servo_tilt and channel_forwarding feature, nothing works (real servos still won't move).

The only way to make them work is set the mixer to AIRPLANE and they suddenly start working - the problem is the AIRPLANE has SMIX defined for servo pins #6 (servo 7) with a throttle, so it's not usable for me.

I tried to do a dump of all configuration when AIRPLANE mixer in use and diff-compare it to CUSTOMAIRPLANE mixer and the only difference is in a name of a mixer and SMIX (SMIX is not listed in the dump when AIRPLANE in use). I tried to do "SMIX LOAD AIRPLANE" when CUSTOMAIRPLANE mixer is in use to get the same configuration for SMIX but with no success, it's still not moving when CUSTOMAIRPLANE mixer is active, but it moves when AIRPLANE mixer is active - when comparing these two dumps, only the name of the mixer is different.

I would say I tried all different combinations, I also used servo tester to make sure all servos are fine, but I can't make it work with CUSTOMAIRPLANE mixer.

TNgineering, 2017-05-03:
Did you get all of your outputs working yet? It looks like your smix lines don't include pins id's 3-5.

Dejw4g, 2017-05-03:
As I wrote, this is a working smix for me:
smix 0 3 0 100 0 0 100 // ailerons
smix 1 4 1 100 0 0 100 // elevator
smix 2 5 2 100 0 0 100 // rudder
smix 3 6 10 100 0 45 100 // flaps
smix 4 7 9 100 0 0 100 // gear

but the problem is, that this:
smix 0 0 1 100 0 0 100 // ailerons
smix 1 1 0 100 0 0 100 // elevator
smix 2 2 2 100 0 0 100 // rudder

and this
smix 0 3 0 100 0 0 100 // ailerons
smix 1 4 1 100 0 0 100 //elevator
smix 2 5 2 100 0 0 100 // rudder

do exactly the same thing - with the same pins used on FC (SP racing F3) - both configurations work for the same pins used on the FC (third, fourth and fifth from left-side)

DzikuVx, 2017-07-04:
OK, thanks for solution. I'm closing this with a remark that there will be changes how servos are handled in Configurator in the future

closed 2017-07-04 by DzikuVx