iNavFlight/inav #1620: REVO only 4 channels work in pwm mode

osergios, 2017-04-30:
As the title says i have an Original Revo board flashed with inav v1.6.1 and i have configured everything and the only issue is that i have only 4 channels working ok in pwm mode. aux1 and aux2 don't see to work and they are connected. 
any clues ?

digitalentity, 2017-04-30:
It would be helpful to have a dump of your configuration.

osergios, 2017-04-30:
# version
# INAV/REVO 1.6.1 Mar  9 2017 / 01:00:33 (6c878ae)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature GPS
feature RX_PARALLEL_PWM
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper GYRO_CALIBRATED
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACTION_SUCCESS
beeper ACTION_FAIL
beeper READY_BEEP
[... 391 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

osergios, 2017-04-30:
Sorry guys found it i had serial rx enabled in ports .
Sorry again and btw very good work quad flies like butter very relaxing keeps altitude better than naza v2 i'm extremely happy.!

digitalentity, 2017-04-30:
Thanks for reporting back, and thanks for using INAV :+1: 
Closing this now.

closed 2017-04-30 by digitalentity