iNavFlight/inav #1623: Possible bug with SP Racing F3 Target

liutairan, 2017-05-01:
I just got a SP Racing F3 board with MPU6050, HMC5883 and MS5611 sensors. I flashed newest iNav 1.6.1 firmware but cannot find Mag and Baro sensors, after I set to the right ones, the configurator still cannot detect the sensors (red icon on sensors). I also tried cleanflight, betaflight, and some old versions of iNavFlight and found out iNav 1.5.1, 1.6.0, 1.6.1 all cannot recognize the HMC5883 and MS5611 sensors. If I set the sensors manually in iNav 1.6.1, then the I2C error will increase continuously and Cycle time will become 20k+, CPU LOAD 600+%.

I am not sure is it the problem with the firmware or just my board or sensors, but as I said previously, it works with the newest cleanflight/betaflight and the old iNav (1.5.0).

digitalentity, 2017-05-01:
Where did you get your SPRacingF3 board? Recently many people are reporting the same issue with these boards.

liutairan, 2017-05-01:
@digitalentity I just saw this issue: https://github.com/iNavFlight/inav/issues/1572 .
I will close this one and go to that issue.

closed 2017-05-01 by liutairan