iNavFlight/inav #1624: Question: Is RTH function viable with inav FW without a MS5611 baro on the FC board ?

mr-sneezy, 2017-05-01:
In more detail. 
If I buy and fit a Ublox GPS module with compass, is the FW going to be able to use RTH to get me home if I loose line of sight ?
Great if so, can fit to the standard SPracingF3 (non deluxe) boards. If not I'd have to replace the whole FC...
Thanks if anybody knows.
Martin

stronnag, 2017-05-01:
Just fit an external BMP280 as well. That's what I'm doing with my SPRF3 acro boards.

mr-sneezy, 2017-05-01:
I could yes, thanks. But is it actually needed, or will RTH function with just GPS Alt (less accurately granted) ?
(I know Alt Hold would work like crap without a baro sensor but I don't use it for FPV flying.)

stronnag, 2017-05-01:
For fixed wing, you don't a baro, for multi-rotor you do.

mr-sneezy, 2017-05-01:
Thanks for confirming, will go with your external BMP280 solution.

cabinw, 2017-05-18:
@stronnag how did you connect external BMP280 to SPRF3 acro boards? via I2C?

skaman82, 2017-05-18:
@cabinw just use the I2C port. Connect SDA and SCL pads and power (5/3v – depending on the module you are using)

cabinw, 2017-05-18:
@skaman82 Got it! Thanks!

closed 2017-05-23 by DzikuVx