iNavFlight/inav #1626: Feature request. Make possible to mount Fc vertical and Mag horizontal

Mow06, 2017-05-02:
1. Off board mag is preferable.
2. The capability is there to mount the FC vertically, on the side of a fuselage for example.
3. A GPS  needs to be mounted horizontal.
4. An external mag on that GPS cannot be used as the mag needs to be in the same plane as the FC.
https://www.rcgroups.com/forums/showpost.php?p=37428051&postcount=13710

Would it be possible to have the capability to use a mag mounted in a different plane to the FC, perhaps a roll option in align_mag?

oleost, 2017-05-02:
Already exsist issues regarding this, please use search function before opening new issue.

#1029 
#86

closed 2017-05-02 by oleost

Mow06, 2017-05-02:
Sorry.Searched but obviously not for the right thing.