iNavFlight/inav #1627: Feature request: stall prevention

brat002, 2017-05-02:
Today my plane was stalled due low airspeed. Mostly, keeping horizon on low speed is not so good. Firmware must allow plane to slightly dive when plane can not keep horizon more. May be will be good decision to decrease Pith force on low speed (at least gps speed if no pitot tube is available)

oleost, 2017-05-02:
There is already low gps speed protection.

Other than that you will need to set your cruise throttle high enough to prevent stall.

brat002, 2017-05-02:
Thank you. Does it anywhere was explained in docs?

oleost, 2017-05-02:
@brat002 Yes, it mention to read CLI document, espesially all marked with "fw_ ".

The First entry for "fw_ " if you search is:

nav_fw_cruise_thr	1400	Cruise throttle in GPS assisted modes, this includes RTH. Should be set high enough to avoid stalling. This values gives INAV a base for throttle when flying straight, and it will increase or decrease throttle based on pitch of airplane and the parameters below. In addiotional it will increase throttle if GPS speed gets below 7m/s ( hardcoded )

brat002, 2017-05-02:
This option related to autonomous mode, but I meant Angle mode. Feature, what I meant, in other words named "stall prevention". Main idea of this feature: prevent to climb altitude on low speeds. In simple case controller should allow the plane decide how to flight level: put nose up when it possible (high speeds) and dive, when altitude can not be keeped due low air speed. On speed, what plane was trimmed, plane should fly straight.
Current behaviour is controller tries to keep level on any speed, what is dangerous on low speeds. To be safe, plane should put nose down and fly in limited range of pitch when throttle is low and up nose in limited range when throttle is high.

Dronek, 2017-05-03:
It might be confusing having a throttle dependent pitch behavior in angle mode.
However,  I like the idea of controlling climb/descend in angle mode by using throttle.
Since throttle stick is usually not spring loaded, the pilot could just apply more than half throttle in order to automatically climb in angle mode without having to hold pitch up (or any other stick) all the time. Similar for descends.

The advantage over climbing/descending in acro mode is that the plane is kept leveled.
As for stall prevention: I think it is not needed in modes where pilot has full control over throttle.

Tdogb, 2017-10-02:
This should really considered. If the airplane is traveling at cruising throttle while traveling with the wind, the airspeed could be very low, and easily allow for a stall. Especially in super windy areas. The groundspeed would be high in this situation but the actual air going over the wings would be low and therefore not generate much lift.

digitalentity, 2017-10-02:
Stall prevention is not possible without an airspeed sensor. Regarding cruise throttle - in a stable flight at cruise throttle it doesn't matter if airplane is traveling into the wind or with the wind - airspeed would be the same in both cases (groundspeed will be very different though). A gust of wind in the tail may cause a stall, but even with an airpseed sensor we won't be able to compensate fast enough - generally it's a matter of properly choosing cruise speed (and cruise throttle).

Airspeed data will eventually be used in automated flight modes for controlling throttle and maintaining safe airspeed in all cases.

Tdogb, 2017-10-02:
Yea, I didn't make that clear, I was thinking of using the airspeed sensor. Also, it does matter if there is a lot of wind traveling in the same direction as the airplane, because when the wind is traveling with an airfoil the speed of the plane and the speed of the wind can cancel each other out, causing there to be extremely slow air moving over and under the airfoil, which will cause the lift of that airfoil to drop dramatically. This affects both he tail of the plane and the wing.
However, I think what you were saying about the airspeed being maintained would fix this problem I am talking about.

closed 2018-05-27 by stale[bot]