iNavFlight/inav #1629: Clipping gyro values in blackbox for omnibusf4

TwoToneEddy, 2017-05-02:
Hi,
I've only just started using black box in inav and noticed the data for the gyros appears to be clipping. I've attached a log file. In this file i don't turn the throttle up at all, i just move the quad in my hands so there is no motor vibration. Any idea what's wrong? It looks like a scaling issue of some kind.
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/971508/LOG00001.TXT)

Klapstuhl, 2017-05-22:
I have the exact same problem with my Omnibus f3 is there any solution to this problem?

Klapstuhl, 2017-05-23:
So i found a solution. If you install the Blackbox viewer app via the Chrome web store the Gyro reading is incorrect but if you install the app manually it works just fine.

DzikuVx, 2017-05-23:
So, problem solved. INAV Blackbox Explorer was never available via Chrome Store, that was CF version maybe? Or BF version? They no longer are compatible. 
The only way to use INAV BB Explorer is via manual installation ATM
Closed

closed 2017-05-23 by DzikuVx