iNavFlight/inav #1632: SP racing F3 board have motors issue on INav1.6~1.7

falconnet88, 2017-05-03:
I tried Inav 1.6 and Inav 1.7, sometimes only 2 of 4 motors can spin up.
In addition, the motors can only be throttle up, no any response to Rud/Aile/Elev.

After I flashed back to Inav1.5, all motors work well...

digitalentity, 2017-05-03:
Please provide config dump and details of your setup.

falconnet88, 2017-05-03:
Sorry, I don't know how to use CLI to check the setup.
These are some screenshots from my current INav configulator setting.

[image]
[image]
[image]

falconnet88, 2017-05-03:
I'm using stock 5" racer setting for my 250mm drone.
Using Blheli14.6 version.

My friend and I have the same issue.

rb1205, 2017-05-03:
What's the point of having 2 kHz oneshot signals to the ESCs if your looptime is 1 kHz?

falconnet88, 2017-05-03:
Re rb1205: Im using the stock setting of INav1.7 for 250mm racer drones. Didn't change anything. The CPU loading become 50%, is it one of the reason my drone behave strangely?

stronnag, 2017-05-03:
Please post a CLI diff.
* Open the CLI tab
* type `diff`
* Paste the output here (in the <> 'insert code' style in the editor).

Note the the racer setting is intended for an Acro mode racer; no nav / sensors.

falconnet88, 2017-05-04:
Thanks Stronnag, I chose 5" GPS setting today, 4 motors work well now. My friend and I all chose 5" racer setting and opened GPS before.

digitalentity, 2017-05-04:
@falconnet88 I believe the issue can be closed now?

falconnet88, 2017-05-04:
Yes, please. Thanks for the reply for solving this issue which caused by my wrong setting.

closed 2017-05-04 by digitalentity