iNavFlight/inav #1634: GNU ARM Embedded Toolchain version

giacomo892, 2017-05-03:
Should we use 6-2017-q1-update (from https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads, and available in various PPA repository) or 6-2016q4?

I've tried both and I have even flown the 2017 one. Compiled size is different.

Are there any negative downsides on using an updated toolchain?

stronnag, 2017-05-03:
Your choice. We don't mandate specific platform compilers and platform specific pages tend to be updated only rarely. 

On Arch, 6.31 works just fine, and iirc, latest Ubuntu 17.04 had a sufficiently recent cross compiler that using a ppa is not necessary.

giacomo892, 2017-05-03:
Ok, thanks. 

6.3.1 is the version that ships with 6-2017-q1-update

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

6-2016q4 is:

arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 6.2.1 20161205 (release) [ARM/embedded-6-branch revision 243739]

I thought that the code was really tested to be compiled with one particular toolchain.

closed 2017-06-06 by DzikuVx