iNavFlight/inav #1635: Magnetometer/Heading PID in configurator

giacomo892, 2017-05-03:
I'm running 1.7 in order to test it, and I noted that Magnetometer/Heading in the PID tab of the configurator does not have any effect in the configuration (just run a diff).

I always wondered about the meaning of that parameter. 

[image]

Is this a bug in the code or only configurator that left that parameter for ancient versions compatibility maybe?

DzikuVx, 2018-01-20:
Just tested, it really looks like a bug. MAG pid not available in dumo or diff

giacomo892, 2018-01-20:
@DzikuVx It might be a leftover in the configurator since I cannot find any meaningful parameter in the CLI.
It might be there from pre INAV era where mag data wasn't fused into the IMU but taken in account separately.

DzikuVx, 2018-01-21:
with #2669 we can close this one I suppose

closed 2018-01-21 by DzikuVx

adri94, 2021-01-20:
Hello,

I have a tricopter LR with a kakuteF7 and tekko32 ESC, the servo is a digital ES3054 from Emax. 

I flashed the FC with iNAV and the tricopter setup, I did the calibration but the heading is drifting one degree at a time every 15 seconds or so, still in the same direction.

If someone can point me in the right direction...

Regards,

PR #2669 (2018-01-21): P gains for Heading Hold in CLI
Looks like those settings were not available in CLI
This fixes #1635