iNavFlight/inav #1637: GPS_PROTO_UBLOX_NEO7PLUS on F3 and F4 targets

rb1205, 2017-05-03:
Looking at the code I found this switch that configures the GPS for better performances, but it's only defined on the AnyF7 target. Is it safe to test on F3 and F4 targets or there's more behind that makes it unflyable? I searched for issues and all I found is #431, bud doesn't seems anything critical.

digitalentity, 2017-05-03:
Unfortunately many cloned M7 and even M8 GPS units don't support PVT message and enabling `GPS_PROTO_UBLOX_NEO7PLUS` makes them useless.

rb1205, 2017-05-03:
I see. Still, is this safe to test? Should I expect better navigation performances if my GPS works with it?

Also if this is the case you may want to remove it from the AnyF7 target as well.

stronnag, 2017-05-03:
It works just fine on my F3 FCs with m8n.It may or may not work for your devices. Your choice.

rb1205, 2017-05-03:
OK, will test and post back any result. I'm closing this.

closed 2017-05-03 by rb1205

rb1205, 2017-05-06:
Tested GPS_PROTO_UBLOX_NEO7PLUS on 1.7 on a 400mm quad. All GPS flight mode worked flawlessy. I noticed a better handling of braking when entering POSHOLD while going at speed. Other than that no noticeable improvements, at least at naked eye.

FC: SP3Evo

GPS: Belitian BN-880