iNavFlight/inav #1638: 1.7 RC1 - Blackbox Viewer issues

DerElch87, 2017-05-03:
Hi there,

I realized two "problems".

1. The motors symbols show all the time 100%. In the value table is the correct motor value.
Version 1.6.1 didn't show that problem.

2. debug mode: gyro or notch aren't shown in the Blackbox viewer. They are simply not chooseable. There is no debug available.
Nav_debug is pickable but shows no values (I thick it is only for nav, or?). Same for 1.7 RC1 and 1.6.1

INav 1.7 RC1 & 1.6.1
FC: SP Racing F3 with removed compass
Enhanced Blackbox Explorer version 2.5.11

stronnag, 2017-05-03:
https://github.com/iNavFlight/blackbox-log-viewer

DerElch87, 2017-05-03:
Sorry I didn't realize that Inav uses a separate Blackbox Viewer.

It is not findable via chrome store and on top via the link you sent me the chrome web store method is linked to the old cleanflight blackbox viewer.
Only the manuel installation way shows that there is a separate blackbox viewer.

Is it probably possible to write that somewhere in the wiki that others also can find it more easily?

stronnag, 2017-05-03:
First it would be helpful to know if that solves your blackbox issues.

Anyone can create a github account and change the wiki, however we should need to change all the documentation, e.g. https://github.com/iNavFlight/inav/blob/master/docs/Blackbox.md

DerElch87, 2017-05-03:
Yes that solved the issues.
Thank you very much for your quick feedback.
I think I need to deepdive a little more into this GitHub things to try to help at least a little.
Is there anywhere sth like a "how to" quickstart?

Last comment:
Wouldn't it be the best thing to make one BB viewer for all? Like using the Enhanced one from Betaflight also for INav?
It has, at least till today, some advantages with the frequencie things like zooming in etc.

stronnag, 2017-05-03:
It's an open source project, contributions are always welcome. 

You may also find that the 1.x series of the *betaflight* blackbox viwer works with iNav, and  even consider raising a bug against that for breaking compatibility ....

DerElch87, 2017-05-03:
OK, I'll raise a bug for the enhanced viewer.
BTW:
I found one last thing. Even the INav viewer shows some problems with some values, see attached screenshot.
Thanks once again.
[image]

closed 2017-05-03 by DerElch87