iNavFlight/inav #1639: Omnibus F3 PPM in only working with UART3 enabled, thus blocking PWM5/6 (required for airplane)

wilco1967, 2017-05-03:
Omnibus F3 basic 1.1 (Banggood), iNav 1.7.0RC1, Airplane mixer, PPM receiver (Frsky D4R-II)

The PPM input is only working when UART3 is enabled (for anything, doesn't matter what)
If nothing is selected for Uart3, PPM input will not work.

However, when enabling Uart3, the PWM5+6 outputs no longer work. Those outputs are required for airplane. With Uart3 disabled, PWM5+6 do work, but PPM input doesn't.....
(On the Omnibus F3, UART3 pins are shared with PWM5+6 pins)

When flashing with betaflight, I don't need to select anything on Uart3 for PPM to work.
So it seems like iNav needlessly requires something to be activated on Uart3 for PPM to work.

I noticed there are two jumpers soldered adjacent to the PPM input, but I cannot find any information what they do, and whether or not I should remove one of them (which one ?)

But in Betaflight, PPM input working just fine with both jumpers installed, and no Uart3 selected, seems to suggest it might be software related.

povlhp, 2017-05-10:
We need 4 PWM outputs for airplane.  I use iBus myself.
Reading the Betalight docs on software serial, they say one problem with iNav is, that they still use the old static resource mappings. And that I2C on PWM 7+8 supposedly is hardwired in, and required. This gives another challenge.

Maybe we could make a new target OMNIBUS-PLANE, without I2C, and use PWM 7+8 as 2x extra servo outputs. That would keep UART 3 free, but sacrifice I2C typically used for external magnetometer which we do not need for plane.

plogibear, 2017-05-22:
I have to agree with wilco1967. Same setup Omnibus F3 basic 1.1 (Banggood), iNav 1.7.0RC1, Airplane mixer, PPM receiver (Frsky D4R-II).

If UART3 is enabled motor outputs PWM 5 & 6 will not work.

To go around I opted to use an SBUS reciever on UART3 only to so that I could get PWM 5 & 6 to work.

Feldsalat, 2017-05-22:
It is necessary to remove the SBUS bridge to use PPM and not enable UART3!

digitalentity, 2017-05-23:
@wilco1967 you need to remove S.Bus jumper to use PPM. It's a hardware issue so I'm closing this.

closed 2017-05-23 by digitalentity

wilco1967, 2017-05-24:
Ok, Thanks !
I'll try removing SBus jumper later.... 

For now, I've got myself a wingwing Z-84, (flying wing) which is working fine with iNav on the Omnibus F3.... 

Problem was, I couldn't find any information regarding the jumpers, and I didn't want to risk removing it, and then finding out this wasn't the solution after all...

Thanks !

Redshifft, 2017-06-08:
What a monumental balls up !!  ...great

a-425, 2018-03-26:
Hi, I have an issue with same board you use. But different Issue. I saw you said : With Uart3 disabled, **_PWM5+6 do work, but PPM input doesn't.....
(On the Omnibus F3, UART3 pins are shared with PWM5+6 pins)_**
how to disable UART3 exactly..???? on port configuration or with CLI to free the pin of UART 3? I use the board to make a quads but I need PWM5 & 6 to run additional servo.

shellixyz, 2018-03-26:
@a-425 You need to remove this link to use PPM and PWM6:

[image]

a-425, 2018-03-27:
Let me try....

jokam85, 2018-07-09:
Any success?

a-425, 2018-07-10:
Yes, but I didn't follow the way upsides. I try with my own way :D